OpenInverterGateway/OpenInverterGateway #186: Questions about uploading to stick

ricter1, 2023-05-13:
## What I'm trying
Uploading a sketch to the stick using Arduino IDE. The sketch should blink LED3 (IO16). But I'm still not sure what to do exactly.
I read the instructions on https://github.com/otti/Growatt_ShineWiFi-S/blob/master/Doc/ShineWiFi-X/ShineWiFi-X.md

## Stick type
ShineWifi-X

## My questions

1. I need to make sure the GND and GPIO0 are connected together while powering up (connecting the USB to pc). I look at the example picture in the docs: https://raw.githubusercontent.com/otti/Growatt_ShineWiFi-S/master/Doc/ShineWiFi-X/ShineWiFi-X_PCB_Bot.jpg 
There I see the pins of the ESP connected together through a 1K resistor. I compared the pins in that image to the pinout diagrams of the ESP-07s on the internet. See this image of the pinout for example: https://community-assets.home-assistant.io/original/4X/9/6/3/963c2d4f287f75e2072293ef073940a70e5f4627.jpeg
When I look at the example picture in the docs, I see the resistor is on the right side of the MCU indeed connected to IO0, but on the left side the resistor doesn't seem to be connected to the GND, but to ADC. I'm confused about this. Am I missing something?

2. Could I use a resistor of another value than 1K? 
3. In the image from the docs, the resistor is soldered. This seems illogical to me, since I assume you have to be able to remove the connection between IO0 and GND after uploading? Or can you just leave it there as long as you don't have a program that suffers from it?

I installed the boards in the Arduino IDE from http://arduino.esp8266.com/stable/package_esp8266com_index.json
A COM port is available in the Arduino IDE.
The CH340 drivers are installed.
When I click "board info" in the Arduino IDE, I get the followng: _BN: Unknown board, VID: 0x1A86, PID: 0x7523, SN: (null)_

I'm asking this question mainly because this is fairly new to me. All I've actually done so far is connect the PCB via USB and see if an available COM port shows up in the Arduino IDE. Maybe some information can be added to the docs.

Here's a picture of mine:
[image]

CarstenCroonenbroeck, 2023-06-29:
I used to suffer from the same confusion. I connected IO0 and what indeed seems to be ADC, and wasn't able to gain write access for the flash tool. Therefore, I also researched the chip's pinout. I ended up connecting IO0 and the actual GND pin (bottom pin on the right) using a simple wire during connect, and just like that, was able to flash. I am now enjoying the otti firmware every day! :-)

crasu, 2023-06-29:
@CarstenCroonenbroeck do we need to update the documentation?

CarstenCroonenbroeck, 2023-06-30:
> @CarstenCroonenbroeck do we need to update the documentation?

I can't tell for sure, because I'm far from being an expert on these chips. However, as I said, the pinout @ricter1 found via Google was the same as I stumbled over, and as it turns out, this is the pinout of the chip package I happened to find in my WiFi stick as well. So at least there should be a note in the documentation hinting this.

crasu, 2023-07-29:
Update the documentation.

closed 2023-07-29 by crasu