OpenInverterGateway/OpenInverterGateway #188: Changing ModBus register value via MQTT command

csc-solution, 2023-05-16:
# What happened?
Is there any possibility to set the register value via MQTT? That would be a very great feature, because it make it possible to automaticly set WorkingMode, etc. with HomeAssistant.

# Stick type
ESP8266

Which Inverter are you using?
SPH 8000TL3 BH-UP

tbasnoopy, 2023-05-17:
Currently this is not possible.

crasu, 2023-05-18:
I do not think that changing registers directly on the inverters via mqtt is really a good idea. However I am open to establish an API that will allow one to change those settings.

closed 2023-05-18 by crasu

csc-solution, 2023-05-18:
An API would be a perfect solution because actually there is no comfortable way to set Register (ex. WorkingMode) over network.