OpenInverterGateway/OpenInverterGateway #191: Growatt config ap still visible

matthiasjost76, 2023-05-19:
# What happened?
Everything is fine. The system is running for 2 months an sending its data to mqtt.

But: why is the growatt config wifi access point still there and visible with the default password.
I can reach the esp from my local network and also I can connect to „growatt config“ and it is reachable at 192.168.4.1. Luckily it does not show my wifi password, but I am afraid that someone can find it out.

# Stick type

What Stick type are you using?
- wemos d1 mini

Which Inverter are you using?
- MIC-600-TL-X

crasu, 2023-05-19:
Did you compile from source or use the precomputed fw?

matthiasjost76, 2023-05-19:
precompiled. Download 23-03-08
[image]

RaptorSDS, 2023-05-24:
have the same problem with not precompile build , build per Platform IO 24.05.2023 and AP still visible after setting up own WIFI (sta)

crasu, 2023-06-04:
That is pretty strange. Does this behaviour persist when you restart the stick? I can only imagine that this happens when the configuration is triggered.

I checked with the latest precompiled version on a lolin32 and it does not happen to me.

RaptorSDS, 2023-06-04:
it happens only when you are get stuck betwen config STA and first real read from Inverter
 , when you get "unknown Shine-Stick" the AP are still visible

 also the first page didnt get update still Manager also on STA Network 

after you solve "unknow Shine Stick " and manage to read from Inverter my AP now not visible anymore

crasu, 2023-06-04:
I tried without an inverter connected. Still I cannot reproduce this. Feel free to reopen this with a detailed description how to reproduce this issue.

closed 2023-06-04 by crasu