OpenInverterGateway/OpenInverterGateway #193: one time data with 600TL-X / missing site

RaptorSDS, 2023-05-24:
# What happened?
only after restart of wemos data are read from inverter after that no json are generate

config.h are 1 for all this but nothing found
/debug site not found 
/firmware site not found 
/uistatus not found 

also please can you add information 600TL are use modbus 1.20 on your readme.md / info field in config.h

# Stick type

What Stick type are you using?
Wemos D1 ch340
Build today with platform io some warning in webdebug 

Which Inverter are you using?
- MIC-600-TL-X
- ....

# Log output

```
Insert logs here
after every restart
```C:\Users\xxxxx>curl -i -X POST http://192.168.2xxx/status -H 'Content-Type:application/json'
HTTP/1.1 200 OK
Content-Type: application/json
Content-Length: 1311
Connection: keep-alive
Keep-Alive: timeout=2000

{"InverterStatus":1,"InputPower":10.5,"PV1Voltage":50.8,"PV1InputPower":10.5,"PV1InputCurrent":0.1,"PV2Voltage":0,"PV2InputPower":0,"PV2InputCurrent":0,"PV3Voltage":0,"PV3InputPower":0,"PV3InputCurrent":0,"PV4Voltage":0,"PV4InputPower":0,"PV4InputCurrent":0,"PV5Voltage":0,"PV5InputPower":0,"PV5InputCurrent":0,"PV6Voltage":0,"PV6InputPower":0,"PV6InputCurrent":0,"PV7Voltage":0,"PV7InputPower":0,"PV7InputCurrent":0,"PV8Voltage":0,"PV8InputPower":0,"PV8InputCurrent":0,"OutputPower":10,"GridFrequency":49.96,"GridL1Voltage":231.4,"GridL1OutputCurrent":0.6,"GridL1OutputPower":11.4,"GridL2Voltage":0,"GridL2OutputCurrent":0,"GridL2OutputPower":0,"GridL3Voltage":0,"GridL3OutputCurrent":0,"GridL3OutputPower":0,"GridRSVoltage":231.4,"GridSTVoltage":0,"GridTRVoltage":0,"EnergyToday":1.7,"EnergyTotal":189.3,"WorkTimeTotal":6199208,"PV1EnergyToday":1.7,"PV1EnergyTotal":192.7,"PV2EnergyToday":0,"PV2EnergyTotal":0,"PV3EnergyToday":0,"PV3EnergyTotal":0,"PV4EnergyToday":0,"PV4EnergyTotal":0,"PV5EnergyToday":0,"PV5EnergyTotal":0,"PV6EnergyToday":0,"PV6EnergyTotal":0,"PV7EnergyToday":0,"PV7EnergyTotal":0,"PV8EnergyToday":0,"PV8EnergyTotal":0,"PVEnergyTotal":192.7,"InverterTemperature":27.6,"InverterIPMTemperature":27.6,"InverterBoostTemperature":0,"BusPVoltage":358.9,"BusNVoltage":0,"RealOutputPowerPercent":0}

after 5sec
C:\Users\xxxx>curl -i -X POST http://192.168.2.xxx/status -H 'Content-Type:application/json'
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 0
Connection: keep-alive
Keep-Alive: timeout=2000

after 15sec
C:\Users\xxxx>curl -i -X POST http://192.168.2.xxx/status -H 'Content-Type:application/json'
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 0
Connection: keep-alive
Keep-Alive: timeout=2000

after 1min
C:\Users\xxxx>curl -i -X POST http://192.168.2.xxxx/status -H 'Content-Type:application/json'
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 0
Connection: keep-alive
Keep-Alive: timeout=2000

RaptorSDS, 2023-05-26:
after Playing with some code and also activate the debug_serial function (i change it to pind D4 serial 1 port) 
now i see all pages

but no data is there , 
debug page say unknown shinestick

but i know its work ( first shot after restart )

crasu, 2023-06-04:
I am happy that you resolved the issue yourself.

> also please can you add information 600TL are use modbus 1.24 on your readme.md / info field in config.h

I think this info is already in the readme [file](https://github.com/otti/Growatt_ShineWiFi-S#supported-inverters):
_Growatt MIC 600-3300TL-X (Protocol 124 via USB/Protocol 120 via Serial)_

closed 2023-06-04 by crasu