OpenInverterGateway/OpenInverterGateway #197: Mod 10ktl3-xh

Derirrepeter, 2023-05-29:
# What happened?
I tried to geht all the Values from the Inverter, i tested al protocols 120 124 and 305 i have valid Values with the 120 protocol. But a Lot of Values are Missing. Whats wrong? So i dont See the consumption of the House and Not Values from the Bat.

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MOD-10ktl-XH

Did anyone manged to get all relevant Values from this Inverter? So pleased let me know.
Thanks a Lot to all af you working in this here

faniskou, 2023-06-04:
One vote here . I have the same issue. Can you help on how we can identify the protocol we have  ? And if it is a new protocol , 
 can you write some more words on how we can add a protocol ? . Thanks any way for the interface contribution

jkoner, 2023-06-08:
I have the MOD-6ktl-xh.
For me it worked with the following fields, when I query them using python. Somehow I did not manage to get them queried by the C++ code. 
Sorry for the different syntax of the field desciption:

offset;field name;length;factor;float/int
3001;ppv;32b;0.1;float
3003;PV1Voltage;16b;0.1;float
3004;PV1Current;16b;0.1;float
3005;ppv1;32b;0.1;float
3007;PV2Voltage;16b;0.1;float
3008;PV2Current;16b;0.1;float
3009;ppv2;32b;0.1;float
3023;pac;32b;0.1;float
3025;GridFrequency;16b;0.01;float
3041;pacToUserTotal;32b;0.1;float
3043;pacToGridTotal;32b;0.1;float
3045;pacToLocalLoad;32b;0.1;float
3051;eacTotal;32b;0.1;float
3053;epvTotal;32b;0.1;float
3067;etoUserToday;32b;0.1;float
3069;etoUserTotal;32b;0.1;float
3071;etoGridToday;32b;0.1;float
3073;etoGridTotal;32b;0.1;float
3075;elocalLoadToday;32b;0.1;float
3077;elocalLoadTotal;32b;0.1;float
3093;InverterTemperature;16b;0.1;float
3094;TemperatureInsideIPM;16b;0.1;float
3095;BoostTemperature;16b;0.1;float
3097;ComTemperature;16b;0.1;float
3169;BatteryVoltage;16b;0.01;int
3171;bmsSoc;16b;1.0;int
3176;bdc1Temp1;16b;0.1;float
3178;bdc1DischargePower;32b;0.1;float
3180;bdc1ChargePower;32b;0.1;float
3182;bdc1DischargeTotal;32b;0.1;float
3184;bdc1ChargeTotal;32b;0.1;float

crasu, 2023-06-29:
Closing this as the issue has been resolved. Feel free to open a pr for the missing values.

closed 2023-06-29 by crasu