OpenInverterGateway/OpenInverterGateway #203: Does it work with ShineLan?

roelarents, 2023-06-09:
# What happened?
Nothing yet ;) But I'm trying to find out if this works with the ShineLan-X stick too? I cannot find anything searching the issues or docs for lan. Maybe they are the same and this is a stupid question, sorry if so.

# Stick type

What Stick type are you using?
- ShineLan-X

Which Inverter are you using?
- MIN-5000 TL-XH

# Log output
n/a

crasu, 2023-06-29:
No we only support wifi.

closed 2023-06-29 by crasu

mwalle, 2023-12-11:
No it is not ESP based, it is based on a STM32. But I've developed an alternative firmware for that stick available here:
https://github.com/mwalle/shinelanx-modbus

Sorry for hijacking such an old issue. @crasu would you mind adding a link to this firmware to the readme? I could make a pull request if you like.