OpenInverterGateway/OpenInverterGateway #204: Error: Unknown Shine Stick - After restart for a longer time

Amenophis86, 2023-06-09:
# What happened?
After I restart the stick or change from "config access point" back I get the error for a longer time. I cant tell how long it takes, but I do not think it is a normal behavior, is it?

# Stick type

What Stick type are you using?
ShineWifi-X 

Which Inverter are you using?
Growatt MIC 600

# Log output
?

```
#0: Setup()
#1: Error: Unknown Shine Stick
#2: Error: Unknown Shine Stick
#3: Error: Unknown Shine Stick
#4: Error: Unknown Shine Stick
#5: Error: Unknown Shine Stick
#6: Error: Unknown Shine Stick
```

Amenophis86, 2023-06-09:
I thought about uploading the firmeware again with OTA but ip/firmware shows: Not found: /firmware

xeno27, 2023-06-23:
if it shows not found /firmware you simply deactivate ota (this happens for me too). the error is normal since it connects to the inverter after some time.

OR

is the stick not connecting to the inverter ?

Amenophis86, 2023-06-24:
The Stick just takes about 1h to Connect with the Inverter. I was wondering why it takes this long. But it works. Sometimes even faster but till it is connected it shows the error in the Log. 

And you are right, I forgot to activate OTA ;)

crasu, 2023-06-29:
@Amenophis86 try to disable serial debugging.

closed 2023-06-29 by crasu