OpenInverterGateway/OpenInverterGateway #212: Setup website is white

majuss, 2023-06-28:
# What happened?
I compiled the latest source with PIO, uploaded via PIO. When the AP is opened I can connect and go to the captive portal. As soon as I click the wifi setup, with or wothout scan, the website just stays blank. See the screenshot.

[image]
 192.168.4.1

Browser and device doesn't matter.

What Stick type are you using?
- Shine X

# Log output
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  GrowattConfig
*wm:[1] AP IP address:
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] Portal Timeout In 300 seconds
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Portal Timeout In 294 seconds
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started 
*wm:[2] WiFi Scan completed in 2282 ms
*wm:[1] 3 networks found
*wm:[2] AP:  -51 Kraftwerk_Messe_2G
*wm:[2] AP:  -51 KraftwerkIoT
*wm:[2] AP:  -51 Kraftwerk_2G
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Portal Timeout In 272 seconds

```
Insert logs here
```

crasu, 2023-06-29:
Which browser are you using the window looks pretty strange? Could you try to connect to the setup ap with firefox/chrome?

majuss, 2023-06-30:
This is the captive window which pops up on macOS. Browser and device doesn't matter, tried Chrome on macOS and Firefox on Android. Everywhere it's the same. I don't have these problems with ESPHome- or other ESP devices.

AditaNi, 2023-06-30:
For me it is working fine. Do you have maybe a dns server which block incoming internet traffic?

majuss, 2023-06-30:
Nope nothing like that, my guess is, the Wifi manager is crashing somehow, because directly after scanning it hangs. Maybe the names of my SSIDs is doing that. I will test it in an envirnoment where no SSIDs are available.

AditaNi, 2023-06-30:
are you using the latest version for the Wifi manager? if yes, try to use one version older.....maybe it's helps.

majuss, 2023-06-30:
I'am compiling the latest source via PIO so     https://github.com/tzapu/WiFiManager#94bb90322bf85de2c9ec592858f20643161bc11f

jrutkin, 2023-06-30:
I have the same issue, blank screens on any device trying to access it, it randomly will work eventually some way some how but the working bit is totally random, mostly it doesn’t..

benjaminnestler, 2023-07-03:
I got the same issue with release binary v1.0.0 [firmware-ShineWifiX.bin](https://github.com/otti/Growatt_ShineWiFi-S/releases/download/release%2Fv1.0.0/firmware-ShineWifiX.bin).

I can transmit my Wifi Credentials by using the "Configure WiFi No Scan" button.

crasu, 2023-07-08:
Please try this PR #218. I could reproduce this on an esp8266 with http updater support enabled. After the fix the issue no longer appears.

majuss, 2023-07-09:
Nope for me the site is still white.

AditaNi, 2023-07-09:
[image]

for me it is working but battery temperature is really wrong!

majuss, 2023-07-10:
@crasu Your latest commit to the PR seems to get it working for me. I will test it today connected to the inverter.

crasu, 2023-07-12:
This seems to be a memory issue in wifiManager. I still get a white page when a lot (10+) wifi networks are detected. I tried to optimize the memory footprint a bit But even with a smaller mqtt send buffer the issue still occurs (sometimes).

closed 2023-07-12 by crasu