OpenInverterGateway/OpenInverterGateway #215: Garbage data sometimes written to MQTT

slovdahl, 2023-06-30:
# What happened?

A few times per day, I get lots of log rows like this in my Home Assistant log:

```
2023-06-28 16:02:40.461 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"InverterStatus":1,"InputPower":3337.2,"PV1Voltage":289.7,"PV1InputCurrent":6.1,"PV1InputPower":1767.1,"PV2Voltage":230.9,"PV2InputCurrent":6.8,"PV2InputPower":1570.1,"OutputPower":3179.5,"GridFrequency":49.96,"L1ThreePhaseGridVoltage":234.9,"L1ThreePhaseGridOutputCurrent":4.5,"L1ThreePhaseGridOutputPower":1057,"L2ThreePhaseGridVoltage":241.2,"L2ThreePhaseGridOutputCurrent":4.5,"L2ThreePhaseGridOutputPower":1085.4,"L3ThreePhaseGridVoltage":232.9,"L3ThreePhaseGridOutputCurrent":4.5,"L3ThreePhaseGridOutputPower":1048,"TodayGenerateEnergy":23.1,"TotalGenerateEnergy":931.6,"TWorkTimeTotal":1556544,"PV1EnergyToday":15.6,"PV1EnergyTotal":517.4,"PV2EnergyToday":7.3,"PV2EnergyTotal":403.8,"PVEnergyTotal":921.2,"InverterTemperature":69.1,"TemperatureInsideIPM":50.2,"BoostTemperature":47.6,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"Ene1\xcf\n\x00\x0cenergy/solar{"InverterStatus":1,"InputPower":3337.4,"PV1Voltage":285,"PV1InputCurrent":6.2,"PV1InputPower":1767,"PV2Voltage":227.6,"PV2InputCurrent":6.9,"PV2InputPower":1570.4,"OutputPower":3176.9,"GridFrequency":49.97,"L1ThreePhaseGridVoltage":234.8,"L1ThreePhaseGridOutputCurrent":4.5,' on energy/solar with encoding utf-8 (for <Job None HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0xffff57f8e980>>)
```

And the JSON from the log output (more) nicely formatted:

```json
{
  "InverterStatus": 1,
  "InputPower": 3337.2,
  "PV1Voltage": 289.7,
  "PV1InputCurrent": 6.1,
  "PV1InputPower": 1767.1,
  "PV2Voltage": 230.9,
  "PV2InputCurrent": 6.8,
  "PV2InputPower": 1570.1,
  "OutputPower": 3179.5,
  "GridFrequency": 49.96,
  "L1ThreePhaseGridVoltage": 234.9,
  "L1ThreePhaseGridOutputCurrent": 4.5,
  "L1ThreePhaseGridOutputPower": 1057,
  "L2ThreePhaseGridVoltage": 241.2,
  "L2ThreePhaseGridOutputCurrent": 4.5,
  "L2ThreePhaseGridOutputPower": 1085.4,
  "L3ThreePhaseGridVoltage": 232.9,
  "L3ThreePhaseGridOutputCurrent": 4.5,
  "L3ThreePhaseGridOutputPower": 1048,
  "TodayGenerateEnergy": 23.1,
  "TotalGenerateEnergy": 931.6,
  "TWorkTimeTotal": 1556544,
  "PV1EnergyToday": 15.6,
  "PV1EnergyTotal": 517.4,
  "PV2EnergyToday": 7.3,
  "PV2EnergyTotal": 403.8,
  "PVEnergyTotal": 921.2,
  "InverterTemperature": 69.1,
  "TemperatureInsideIPM": 50.2,
  "BoostTemperature": 47.6,
  "DischargePower": 0,
  "ChargePower": 0,
  "BatteryVoltage": 0,
  "SOC": 0,
  "ACPowerToUser": 0,
  "ACPowerToUserTotal": 0,
  "ACPowerToGrid": 0,
  "ACPowerToGridTotal": 0,
  "INVPowerToLocalLoad": 0,
  "INVPowerToLocalLoadTotal": 0,
  "BatteryTemperature": 0,
  "BatteryState": 0,
  "EnergyToUserToday": 0,
  "Ene1\xcf\n\x00\x0cenergy/solar{"InverterStatus":1,"InputPower":3337.4,"PV1Voltage":285,"PV1InputCurrent":6.2,"PV1InputPower":1767,"PV2Voltage":227.6,"PV2InputCurrent":6.9,"PV2InputPower":1570.4,"OutputPower":3176.9,"GridFrequency":49.97,"L1ThreePhaseGridVoltage":234.8,"L1ThreePhaseGridOutputCurrent":4.5,'
```

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MOD-7000-TL3-X

# Log output

slovdahl, 2023-06-30:
I digged through the code and I couldn't find any obvious defects. I found https://github.com/knolleary/pubsubclient/issues/489 that sounds like a very similar problem.

> Never mind, it seems that my issue was because I was somehow publishing in the callback... I didn't realise at first that I was publishing in a sub function within the callback ( I knew about the publish_in_callback example...)

I'm not yet familiar enough with this code base, but I don't think this applies in this case? 🤔

crasu, 2023-07-12:
Looks like an overflow of the json output buffer to me. The buffer is 2048 bytes long but the ok looking part of the string has about 1050 bytes. Do you have any other examples?

slovdahl, 2023-07-12:
```
2023-07-07 21:15:16.636 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"InverterStatus":1,"InputPower":624.1,"PV1Voltage":284.8,"PV1InputCurrent":0.3,"PV1InputPower":85.4,"PV2Voltage":239.9,"PV2InputCurrent":2.1,"PV2InputPower":503.7,"OutputPower":614.8,"GridFrequency":50.01,"L1Th
reePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent":1,"L1ThreePhaseGridOutputPower":235.5,"L2ThreePhaseGridVoltage":241.8,"L2ThreePhaseGridOutputCurrent":1,"L2ThreePhaseGridOutputPower":241.8,"L3ThreePhaseGridVoltage":233.9,"L3ThreePhaseGridOutputCurrent":1,"L3ThreePhaseGridOutputPower":233.9,"TodayGenerateEne
rgy":34.6,"TotalGenerateEnergy":1212.6,"TWorkTimeTotal":2237310,"PV1EnergyToday":19,"PV1EnergyTotal":671.2,"PV2EnergyToday":15.4,"PV2EnergyTotal":526.7,"PVEnergyTotal":1197.9,"InverterTemperature":58.6,"TemperatureInsideIPM":32.6,"BoostTemperature":29.7,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"A
CPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"EnergyToUser1\xc6\n\x00\x0cenergy/solar{"InverterStatus":1,"InputPower":622,"PV1Voltage":280.5,"PV1InputCurrent":0.3,"PV1I
nputPower":84.1,"PV2Voltage":239.5,"PV2InputCurrent":2.1,"PV2InputPower":502.9,"OutputPower":612.7,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent' on energy/solar with encoding utf-8 (for <Job None HassJobType.Callback <function MqttAvailability._availability_prepare_subscribe_to
pics.<locals>.availability_message_received at 0xffff8a91eca0>>)
2023-07-07 21:15:16.643 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"InverterStatus":1,"InputPower":624.1,"PV1Voltage":284.8,"PV1InputCurrent":0.3,"PV1InputPower":85.4,"PV2Voltage":239.9,"PV2InputCurrent":2.1,"PV2InputPower":503.7,"OutputPower":614.8,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent":1,"L1ThreePhaseGridOutputPower":235.5,"L2ThreePhaseGridVoltage":241.8,"L2ThreePhaseGridOutputCurrent":1,"L2ThreePhaseGridOutputPower":241.8,"L3ThreePhaseGridVoltage":233.9,"L3ThreePhaseGridOutputCurrent":1,"L3ThreePhaseGridOutputPower":233.9,"TodayGenerateEnergy":34.6,"TotalGenerateEnergy":1212.6,"TWorkTimeTotal":2237310,"PV1EnergyToday":19,"PV1EnergyTotal":671.2,"PV2EnergyToday":15.4,"PV2EnergyTotal":526.7,"PVEnergyTotal":1197.9,"InverterTemperature":58.6,"TemperatureInsideIPM":32.6,"BoostTemperature":29.7,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"EnergyToUser1\xc6\n\x00\x0cenergy/solar{"InverterStatus":1,"InputPower":622,"PV1Voltage":280.5,"PV1InputCurrent":0.3,"PV1InputPower":84.1,"PV2Voltage":239.5,"PV2InputCurrent":2.1,"PV2InputPower":502.9,"OutputPower":612.7,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent' on energy/solar with encoding utf-8 (for <Job None HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0xffff8a91f6a0>>)
2023-07-07 21:15:16.645 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"InverterStatus":1,"InputPower":624.1,"PV1Voltage":284.8,"PV1InputCurrent":0.3,"PV1InputPower":85.4,"PV2Voltage":239.9,"PV2InputCurrent":2.1,"PV2InputPower":503.7,"OutputPower":614.8,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent":1,"L1ThreePhaseGridOutputPower":235.5,"L2ThreePhaseGridVoltage":241.8,"L2ThreePhaseGridOutputCurrent":1,"L2ThreePhaseGridOutputPower":241.8,"L3ThreePhaseGridVoltage":233.9,"L3ThreePhaseGridOutputCurrent":1,"L3ThreePhaseGridOutputPower":233.9,"TodayGenerateEnergy":34.6,"TotalGenerateEnergy":1212.6,"TWorkTimeTotal":2237310,"PV1EnergyToday":19,"PV1EnergyTotal":671.2,"PV2EnergyToday":15.4,"PV2EnergyTotal":526.7,"PVEnergyTotal":1197.9,"InverterTemperature":58.6,"TemperatureInsideIPM":32.6,"BoostTemperature":29.7,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"EnergyToUser1\xc6\n\x00\x0cenergy/solar{"InverterStatus":1,"InputPower":622,"PV1Voltage":280.5,"PV1InputCurrent":0.3,"PV1InputPower":84.1,"PV2Voltage":239.5,"PV2InputCurrent":2.1,"PV2InputPower":502.9,"OutputPower":612.7,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent' on energy/solar with encoding utf-8 (for <Job None HassJobType.Callback <function MqttAvailability._availability_prepare_subscribe_topics.<locals>.availability_message_received at 0xffff8a91fba0>>)
2023-07-07 21:15:16.647 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"InverterStatus":1,"InputPower":624.1,"PV1Voltage":284.8,"PV1InputCurrent":0.3,"PV1InputPower":85.4,"PV2Voltage":239.9,"PV2InputCurrent":2.1,"PV2InputPower":503.7,"OutputPower":614.8,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":235.5,"L1ThreePhaseGridOutputCurrent":1,"L1ThreePhaseGridOutputPower":235.5,"L2ThreePhaseGridVoltage":241.8,"L2ThreePhaseGridOutputCurrent":1,"L2ThreePhaseGridOutputPower":241.8,"L3ThreePhaseGridVoltage":233.9,"L3ThreeP [... cut ...] 0,"EnergyToUserToday":0,"Energy1\xd3\n\x00\x0cenergy/solar{"InverterStatus":1,"InputPower":5518.6,"PV1Voltage":263.6,"PV1InputCurrent":12.5,"PV1InputPower":3295,"PV2Voltage":226.9,"PV2InputCurrent":9.8,"PV2InputPower":2223.6,"OutputPower":5302.8,"GridFrequency":49.93,"L1ThreePhaseGridVoltage":237.2,"L1ThreePhaseGridOutputCurren' on energy/solar with encoding utf-8 (for <Job None HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0xffff8a94f560>>)
```

crasu, 2023-07-15:
Could you try this branch? #223

slovdahl, 2023-07-15:
> Could you try this branch? #223

Sure! 👍 I just installed it. I'll let it run for a few days and keep an eye on the HA log.

slovdahl, 2023-07-18:
Interesting, got another type of error now 👇 I _think_ there's an extra `1` at the end of the JSON?

```
2023-07-18 13:41:28.339 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: {"InverterStatus":1,"InputPower":2941.6,"PV1Voltage":306.9,"PV1InputCurrent":5.6,"PV1InputPower":1718.6,"PV2Voltage":249.6,"PV2InputCurrent":4.9,"PV2InputPower":1223,"OutputPower":2812.9,"GridFrequency":50,"L1ThreePhaseGridVoltage":235.1,"L1ThreePhaseGridOutputCurrent":3.9,"L1ThreePhaseGridOutputPower":916.8,"L2ThreePhaseGridVoltage":241.3,"L2ThreePhaseGridOutputCurrent":4,"L2ThreePhaseGridOutputPower":965.2,"L3ThreePhaseGridVoltage":233.6,"L3ThreePhaseGridOutputCurrent":4,"L3ThreePhaseGridOutputPower":934.4,"TodayGenerateEnergy":6.1,"TotalGenerateEnergy":1642.9,"TWorkTimeTotal":2971130,"PV1EnergyToday":3.5,"PV1EnergyTotal":911.5,"PV2EnergyToday":2.4,"PV2EnergyTotal":711.9,"PVEnergyTotal":1623.4,"InverterTemperature":55,"TemperatureInsideIPM":31.1,"BoostTemperature":28.8,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"EnergyToUserTtal":0,"EnergyToGridToday":0,"EnergyToGridTotal":0,"DischargeEnergyToday":0,"DischargeEnergyTotal":0,"ChargeEnergyToday":0,"ChargeEnergyTotal":0,"LocalLoadEnergyToday":0,"LocalLoadEnergyTotal":0,"ACChargeEnergyToday":0,"ACChargeEnergyTotal":0,"Mac":"48:55:19:62:9C:0E","Cnt":8692}1, template: {{ value_json.InverterStatus }})
```

slovdahl, 2023-07-18:
FWIW, the version of #223 I built it from is a2c9c975fefb25d75190ecec12cf5fb6bf8def7d.

crasu, 2023-07-29:
@slovdahl I tried to reproduce this today with the version on my branch. Even when the json gets to long for its buffer. The json serialization lib handles this and outputs correct json (although with missing values). Could you add some logging ShineMqtt::mqttPublish to log measureJson(doc) and the serialized json. You should be abled to log this to a local file why telnet (after setting ENABLE_TELNET_DEBUG).

slovdahl, 2023-08-14:
Annoying.. Telnet stops working each night (when solar productions goes down to 0 I presume?), so I need to restart the telnet session that logs the output to a file each day. And yesterday it happened according to the HA log but I had forgotten to restart the telnet logging 🙈

crasu, 2023-10-04:
I checked my files today and I have not seen this message once. Maybe this is because I am using a single phase inverter. Closing this as it is not reproducible. Feel free to reopen.

closed 2023-10-04 by crasu

slovdahl, 2023-10-10:
:+1: 

The WiFi signal is quite weak where my inverter is placed, so that could also be an explanation of why it happens. I'll get myself another AP at some point to improve the signal, so if I just remember it I can report back when I have it in place.

slovdahl, 2024-06-07:
FWIW, I've got a WiFi AP close to the inverter and Shine-X now, but I still got similar errors in the log. But I have restarted Home Assistant more than twice since I saw it in the log, so the actual log entry is gone for now. It was running the same firmware as last fall at the time, but I have now flashed it with 09bb5fbacb679adb68c637950ce2e88e5abe632b. I'll keep an eye out for any errors going forward.

slovdahl, 2024-06-12:
Got it again yesterday morning.

```
2024-06-11 07:38:58.047 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"InverterStatus":1,"InputPower":1205.3,"PV1Voltage":310.1,"PV1InputCurrent":2.2,"PV1InputPower":682.2,"PV2Voltage":249.1,"PV2InputCurrent":2.1,"PV2InputPower":523.1,"OutputPower":1133.2,"GridFrequency":50.06,"L1ThreePhaseGridVoltage":232.9,"L1ThreePhaseGridOutputCurrent":1.7,"L1ThreePhaseGridOutputPower":395.9,"L2ThreePhaseGridVoltage":239.2,"L2ThreePhaseGridOutputCurrent":1.7,"L2ThreePhaseGridOutputPower":406.6,"L3ThreePhaseGridVoltage":231.3,"L3ThreePhaseGridOutputCurrent":1.7,"L3ThreePhaseGridOutputPower":393.2,"TodayGenerateEnergy":0.8,"TotalGenerateEnergy":6044.8,"TWorkTimeTotal":1.3817344e7,"PV1EnergyToday":0.3,"PV1EnergyTotal":3387.1,"PV2EnergyToday":0.2,"PV2EnergyTotal":2538.3,"PVEnergyTotal":5925.4,"InverterTemperature":48.3,"TemperatureInsideIPM":21.8,"BoostTemperature":20.1,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0EnergyToUserTotal":0,"EnergyToGridToday":0,"EnergyToGridTotal":0,"DischargeEnergyToday":0,"DischargeEnergyTotal":0,"ChargeEnergyToday":0,"ChargeEnergyTotal":0,"LocalLoadEnergyToday":0,"LocalLoadEnergyTotal":0,"ACChargeEnergyToday":0,"ACChargeEnergyTotal":0,"Mac":"48:55:19:62:9C:0E","Cnt":3460}\xc0\x00' on energy/solar with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<bound method MqttEntity._message_callback of <entity sensor.growatt_pv1_input_power=697.8>>, <bound method MqttAvailabilityMixin._availability_message_received of <entity sensor.growatt_pv1_input_power=697.8>>, {'available'})>)
```