OpenInverterGateway/OpenInverterGateway #216: Ethernet support

csc-solution, 2023-07-02:
Are there plans to implement the Ethernet library? With this you could get the inverter with an EthernetShield (e.g. W5500) into the network even without WiFi - that would be a plus for security and network stability.

I have successfully tested it with a D1 mini and a web server

crasu, 2023-07-12:
I am open to PRs, but there are currently no plans. 

The shine LAN-X is not supported at the moment, but I am open to pr there as well:
https://de.growatt.com/upload/file/20221121/9694510375581ba05053a5b4e5aede79.pdf

closed 2023-07-12 by crasu

csc-solution, 2023-07-12:
Ok, sounds good. What can I do to help you with the development? Do you need a LAN-X stick and information on how to flash it?