OpenInverterGateway/OpenInverterGateway #225: Support for multiple inverters - Paralleled / 3 phases?

RoganDawes, 2023-07-20:
# What happened?
I am looking to build a solution for a friend who has 5 Growatt inverters (not 100% sure of model), supporting a 3 phase supply. Two phases have two inverters paralleled to support the load. Before I go too far down the path, I'm trying to figure out how many ESP32's I need. I could see the answer being 5, 3 or 1, depending on how the inverters are connected to each other, and whether this project is able to query other inverters across e.g. RS485 or other connections.

I'm also wondering whether it is possible to use an external CH340-based USB-TTL adapter wired up to the second and third UART on the ESP32 to talk to the additional inverters. Not seeing anything in this regard on initial look, but wondering whether there are any fundamental reasons why this should be possible (with code changes, if necessary)?

# Stick type

What Stick type are you using?
- Other (generic ESP32 with CH340 USB-serial chip)

Which Inverter are you using?
- not 100% sure, but has USB interface to the dongles.

# Log output
N/A

crasu, 2023-07-20:
You need one esp per inverter. More than one inverter is not supported by the fimrware.

crasu, 2023-07-20:
Closing as not a bug.

closed 2023-07-20 by crasu

RoganDawes, 2023-07-20:
Would you accept contributions to allow querying multiple inverters using usb-ttl converters?

crasu, 2023-07-20:
Sure - contributions are always welcome.

RoganDawes, 2023-08-10:
Has anyone reported success with the SPF 5000TL inverters? Protocol doc is here: https://watts247.com/manuals/gw/GrowattModBusProtocol.pdf

I guess I will have to try different slave addresses to see whether I am actually getting responses from the inverters.