OpenInverterGateway/OpenInverterGateway #227: No data for HomeAsistant

AditaNi, 2023-07-29:
# What happened?
after flashing the latest SW release I see that the stick ist not always recognised.
After a while of waiting the stick it is detected but no data is to be seen in homeassistant.
Mqtt connection is established.
On the main page the values are shown. but not data for homeassistant.

for example:
mqtt:
  sensor:
    - name: "InverterStatus"
      state_topic: "energy/solar/growatt"
      unique_id: "InverterStatus"
      unit_of_measurement: ""
      value_template: '{{ float(value_json.InverterStatus)| round(1) }}'
returns now:
UndefinedError: 'value_json' is undefined -> Why?

# Stick type

What Stick type are you using?
- Shine X
- Shine S -> this stick I have.
- Lolin32
- Nodemcu32
- Other 

Which Inverter are you using?
- SPH-4000 BL-UP

# Log output
[image]

```
Insert logs here
```

AditaNi, 2023-07-29:
Data on main page!

[image]

crasu, 2023-07-29:
Could you check how the mqtt messages are formatted for example with mosquitto_sub?

AditaNi, 2023-07-29:
with the last version it was working fine .

please let me know how to check the mqtt.

crasu, 2023-07-30:
The easiest way is probably to just copy the unparseable messages from your home assistant logs:
https://www.home-assistant.io/integrations/system_log/

I am using this comand on linux cli:
`mosquitto_sub -h homeassistant -t "#"`

There seems to be a homeassistant addon as well:
https://community.home-assistant.io/t/mqtt-explorer-addon/465758

AditaNi, 2023-07-30:
Thanks for your Informations.
I have change the mqtt protocol from 3.1.1 to 3.1 and it is working again.
🙏

closed 2023-07-30 by crasu

lukcob, 2023-10-05:
Hey guys,

I have the same problem. Where do you change the mqtt protocol? And what is your configuration at the yaml Code?