OpenInverterGateway/OpenInverterGateway #234: Multiple "ReadData() NOT successful"

sveba, 2023-08-10:
# What happened?
Multiple "ReadData() NOT successful" with Wemos D1 mini - ESP8266 
May be this is OK!? 
I compiled by myself with 15s REFRESH_TIMER and GROWATT_MODBUS_VERSION 120 (124 doesn't work)

# Stick type
Wemos D1 mini - ESP8266 (USB with CH340)

Which Inverter are you using?
- MIN-4600-TL-X
- 
# Log output

```
ReadData() NOT successful
ReadData() NOT successful
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() NOT successful
ReadData() successful
```

crasu, 2023-08-13:
This can happen as long as serial debugging is enabled. Turning of serial debugging should resolve this.

closed 2023-08-13 by crasu

sveba, 2023-08-13:
Looks good without serial debugging. Still 10% `ReadData() NOT successful` but much better.
Thanks