OpenInverterGateway/OpenInverterGateway #235: Shine wifi-x

reinoso360, 2023-08-12:
# What happened?
Hello everyone, I have two shine wifi-x that after updating with the growatt firmware 3.1.0.5 got stuck with 3 leds on straight and it doesn't work anymore. I put a jumper between the GPIO0 pin 12 of the esp, and connected it to the gnd, before putting it in the computer's usb. So he agrees to download it in any way. What am I doing wrong, or is there a possibility that this firmware from growatt has stopped esp?

# Stick type
Shine wifi-x

Which Inverter are you using?
- MIC-2000-TL-X

# Log output
esptool.py v4.7-dev
Found 1 serial ports
Serial port COM8
Connecting...
COM8 failed to connect: Write timeout

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.

```
Insert logs here
```

crasu, 2023-08-13:
Thank you for reporting the issue. I need the serial log to investigate this. Could you please post it.

reinoso360, 2023-08-13:
Hello crasu, how to I get the log? I've tried to use Flash tools, and esptool to download the bin, but it doesn't bring any log. 
[image]

crasu, 2023-08-13:
Here is a description:
https://learn.adafruit.com/adafruit-esp32-s2-tft-feather/advanced-serial-console-on-windows

reinoso360, 2023-08-28:
Hello again, crasu, I bought a new esp8266 dev board and install the firmware, I have made the config and it connect to my wifi, but I don't know how can I connect to a mqtt and save the data. Please help, there is a tutorial that I cannot see?

reinoso360, 2023-09-04:
Hello crasu, I bought an ch340 programmer and I have this messages, but cannot download the firmware on the stick. I have tested with an external power supply.

C:\esptool>esptool.py flash_id
esptool.py v4.7-dev
Found 2 serial ports
Serial port COM7
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 24:a1:60:2c:62:c8
Uploading stub...

C:\esptool>esptool.py --baud 9600 --port COM7 read_flash 0x0 0x400000 fw-backup-4M.bin
esptool.py v4.7-dev
Serial port COM7
Connecting..............
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
WARNING: Detected crystal freq 157.29MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 24:a1:60:2c:62:c8
Uploading stub...

crasu, 2023-09-04:
> Hello again, crasu, I bought a new esp8266 dev board and install the firmware, I have made the config and it connect to my wifi, but I don't know how can I connect to a mqtt and save the data. Please help, there is a tutorial that I cannot see?

Have a look at this issue #238 on how to setup mqtt

crasu, 2023-09-04:
> C:\esptool>esptool.py flash_id esptool.py v4.7-dev Found 2 serial ports Serial port COM7 Connecting.... Detecting chip type... Unsupported detection protocol, switching and trying again... [...]

These reading are very strange maybe your stick is broken.

crasu, 2023-09-04:
Closing this as it is not a general issue with the firmware.

closed 2023-09-04 by crasu