OpenInverterGateway/OpenInverterGateway #240: BatteryTemperature wrong steps

WiedenhoeverD, 2023-09-07:
# What happened?
Last week I tested a bit with the repo and my GrowattInverter. It works really great. I noticed that inside the `Growatt124.cpp`  the BatteryTemperature is set up incorrectly. In [this](https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf) document it is also written that the BatteryTemperature is returned in 0.1°C steps.  
[image]  
It is just inside the  `Growatt124.cpp`  at BatteryTemperature. Every other temperature is right with a 0.1°C steps.  
Inside the Website a BatterTemperature with 300°C is very unlikely.  
[image]

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- Growatt SPH 10000TL3 BH-UP

crasu, 2023-09-07:
This seems to be the same issue https://github.com/otti/Growatt_ShineWiFi-S/pull/166

WiedenhoeverD, 2023-09-07:
After reading #166 I also checked my Modbus FW version. It is also 305. I have the same Inverter as @tommarek and we both have a 0.1 scale.

tommarek, 2023-09-07:
I think the possible solution is to have two very similar configurations for both inverter variations. Will create a PR once I get home from holidays.

crasu, 2023-09-09:
May be we need to have a config parameter in config.h for this. I would default to the documented value and have a define to overwrite this.

crasu, 2023-10-07:
@tommarek, @WiedenhoeverD could you try the workaround on this branch #256. Thank you in advance!

PR #256 (2023-10-18): Workaround for different multiplier on same sph inverter firmware version
fixes #240
closed 2023-10-18 by crasu