OpenInverterGateway/OpenInverterGateway #245: Inverter no respoding

OmarGalsoolker, 2023-09-21:
I erased my new ESP8266MOD stick and then uploaded the newly created firmware. After configuring the WiFi and setting up the MQTT server details, I plug in my stick into my inverter (SPF5000ES). The blue light is on.
What am I not doing?

# Stick type
d1:
ESP8266MOD CH340G

My Inverter is:
Growatt SPF5000ES

# Log output
mqtt log: I get the following:
1695332788: mosquitto version 2.0.17 starting
1695332788: Config loaded from C:\Program Files\mosquitto/mosquitto.conf.
1695332788: Opening ipv4 listen socket on port 1883.
1695332788: mosquitto version 2.0.17 running
1695332789: New connection from 192.168.50.184:57613 on port 1883.
1695332789: New client connected from 192.168.50.184:57613 as Growatt3866781 (p2, c1, k15).
1695332789: Will message specified (25 bytes) (r1, q1).
1695332789: 	energy/power
1695332789: Sending CONNACK to Growatt3866781 (0, 0)
1695332789: Received SUBSCRIBE from Growatt3866781
1695332789: 	energy/power/command/# (QoS 1)
1695332789: Growatt3866781 1 energy/power/command/#
1695332789: Sending SUBACK to Growatt3866781
1695332804: Received PINGREQ from Growatt3866781
1695332804: Sending PINGRESP to Growatt3866781
1695332819: Received PINGREQ from Growatt3866781
1695332819: Sending PINGRESP to Growatt3866781
1695332834: Received PINGREQ from Growatt3866781
1695332834: Sending PINGRESP to Growatt3866781
1695332849: Received PINGREQ from Growatt3866781
1695332849: Sending PINGRESP to Growatt3866781
1695332864: Received PINGREQ from Growatt3866781
1695332864: Sending PINGRESP to Growatt3866781
```
Insert logs here
```

crasu, 2023-09-22:
Could you provide the stick log as well. You can get it on the webinterface of the stick (<stick_ip>:8080).

OmarGalsoolker, 2023-09-25:
Apologies for the delayed response: See stick log below:
[image]

crasu, 2023-09-26:
Its a bit strange that it needs 6 attempts to connect to your mqtt server. You can try some things:

1. Sometimes it helps to wait a couple of minutes until the inverter connects
2. Some people had difficulties with the bit rate auto detection. You could hardcode the bit rate.
3. Make sure ENABLE_SERIAL_DEBUG is turned of - it sometimes interferes with the inverter communication
4. For this inverter type you have to build your own firmware with the right protocol version (5000).

OmarGalsoolker, 2023-09-26:
Thanks for this information.
So the reason why there were so many attempts on the mqtt server is that it was not started on the computer. I started it up and then the connection happened almost immediately.
1) I did wait for a while but nothing changed
2) How do I decide which bit rate to use? There are 2 sticks i.e. ShineWiFi_X = 115200 and ShineWiFi_S = 9600
3) ENABLE_SERIAL_DEBUG  was turned off
4) I am a total amature in this area and have no clue how to do this. Would someone be able to assiste me with this?

FYI, As mentioned above, I am an amature and thought i could just compile everything and load it on stick. But it seems that i have to write my own firmware. I also dont know if i implemented evrything correctly. I followed the steps but get the issues above. The "Unknown Shine Stick" error above is telling that the inverter is not reckognizing the stick.
Please guide me through the implemetation one step at a time. I have the following questions in the meantime:
1) Do i need a growatt wifi stick (for my inverter which is SPF5000ES) which i need to modify/change?
2) Do i plug the ESP8266MOD card into my USB on the inverter?
3) Do i need to run an rs485 cabkle to my board and is that the way things work?

I have so many questions PLEASE assist

OmarGalsoolker, 2023-09-26:
Has anybody written firmware for my inverter type i.e. SPF5000ES - please share with me. I will appreciate it.

crasu, 2023-10-04:
Sry I did not get that you are using an SPF inverter. The SPF inverter is not supported. However there is support in the code here:
https://github.com/otti/Growatt_ShineWiFi-S/commit/296938c565b9d4e2d5a15e304011195623e0d604

The support was not written by me but by @nuclearcat. As you can read here it needs a custom adapter:
https://github.com/otti/Growatt_ShineWiFi-S/pull/109

If you are nice to @nuclearcat he might even surrender the drawing he promised half a year ago ;-).

Closing this as ther is not official SPF support.

closed 2023-10-04 by crasu