OpenInverterGateway/OpenInverterGateway #246: PVPower / PEX1 +2? Crazy value if PVPower = 0

hugocrank, 2023-09-27:
# What happened?
When reading the 1st register for InputPower I get values like: 429496736
In the appendix my definition for reading out according to the "old" version of Otti.
In addition, another original SDM630 from Growatt is connected to 485-1.
The condition occurs only when NO power comes!
as soon as the value > 0 on the strings is (Positive, power is generated)
Are the values correct!

Can someone help me either to take this into account in attached code or to "remove / filter" the values?

thanks a lot!

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- SPH-3000 BL-UP
- ....

#Code 
    if (res == Modbus.ku8MBSuccess)
    {
      _Data.u32PVPower         = (Modbus.getResponseBuffer(1) << 16) + Modbus.getResponseBuffer(2);     // PVPower + PEX1 - ???

[image]

hugocrank, 2023-09-27:
Alternatively:
does anyone know the registers for PEX1 when a 2nd inverter is connected? i can't find any documentation on this.

hugocrank, 2023-09-27:
Register for TLX:
[image]

Where to find for SPH?

hugocrank, 2023-09-28:
Register for PEX = 

1131. Extra AC Power
to grid_H
Extra inverte AC Power to grid
High
For SPA
connect
inverter
SPA used
1132. Extra AC Power
to grid_L
Extrainverte AC Power to grid Low SPA used

hugocrank, 2023-09-29:
the problem of the "Big" Value seems to be negative values?  how can i make a workarround for this?

DutchKillerbee, 2023-10-04:
Yeah I noticed this as well. 
I "fixed" this in Home assistant but needs to be fixed in the code.
Unfortunately I don't have the skills
DKB

crasu, 2023-10-04:
@hugocrank where is the screenshot coming from? It seems that you are using a really old firmware version.

I am a bit confused which values you are talking about. 1131+1132 do not exported by current firmware versions.

hugocrank, 2023-10-05:
@dutchkillerbee: thank you for confirming the condition. Since I have built my "own" firmware with my borrowed knowledge, I initially assumed an error on my part. (My firmware was already installed several times due to the larger number of evaluated registers).

But I can now say that the "Grot" server which I also use these values outputs and exactly the same to the Growattcloud outputs in the data stream! This corrects Growatt apparently also downstream in the data preparation.

The registers for external power of Meter2 behave normally for Day/Total! While "Power External" also outputs such values in the absence of PV power of the 2nd meter! It seems as if the problem refers purely to the power of the Meter2! the register PVPower is only summed from DC1Power + DC2Power + Extra AC Power.

Unless this post is "merged" again with other topics or is closed I will gladly share the result here.

DutchKillerbee, 2023-10-06:
FYI 
As far as I can remember I have seen these high numbers on the PVOutput using the latest source code from github.
[image]
They are probably caused by reading a negative value from the inverter. So maybe a conversion error/overflow?
I can remember that I could see negative values (-2W) in HA for the PVOutput when using the original Growatt firmware.
I always wondered why these values were negative.
Maybe these negative values are converted in these extremely high numbers?

PS
I am using modbus protocol version 1.24

hugocrank, 2023-10-07:
You are right! I See this on a sma inverter on meter 2 and with 2 sph, second on meter 2. I habe a conversation accual with growatt and will discusse this. Soltuion seems so be erase negative values in customfirmware bevor they come in a second System.Media-Contest3. Tongraben 1298617 MeiningenT.: +49 3693/471997M.: ***@***.*** Schlotthauer

crasu, 2023-10-07:
> .... Since I have built my "own" firmware with my borrowed knowledge, I initially assumed an error on my part. (My firmware was already installed several times due to the larger number of evaluated registers).

@hugocrank 
You are probably reading a signed register as an unsigned register for signed registers you have to set the type to  SIZE_16BIT_S or SIZE_32BIT_S. As you seem to be using a patched version of the firmware I am unable fix this issue.

closed 2023-10-07 by crasu

DutchKillerbee, 2023-10-08:
@crasu 
As I said in my previous post the issue also occurred with the current (unpatched) firmware.
Maybe you can check the register type of the PVOutput? It needs to be a signed register.
I don;t have the knowledge to check this...

crasu, 2023-10-08:
@DutchKillerbee I am sorry but  there does not seem to be a register with the name "PVOutput". 

Could you open a new issue for this? Please fill out the bug report completly including the protocol and firmware version you are using and the exact name of the register.

hugocrank, 2023-10-09:
Register 1 my hero.

DutchKillerbee, 2023-10-13:
[image]
[image]
What kind of type should I try here to get a signed 32 Bit Integer?
Growatt124.cpp:
[image]