OpenInverterGateway/OpenInverterGateway #247: Battery values stay on 0

WaizenErnter, 2023-09-30:
# What happened?
The SOC, charge and discharge stay on 0.  All the other battery parameters also stay on 0, but i dont care about them. 

What Stick type are you using?
- Shine X

Which Inverter are you using?
-MOD 8000TL3 XH
-ARK XH 10kWh
[image]
[image]

sidequest-tomas, 2023-10-10:
same here, SOC and battery values do not work:

What Stick type are you using?

ESP8266 D1 mini

Which Inverter are you using?

-MOD 10000TL3 XH
-ARK XH 10kWh

{"InverterStatus":1,"InputPower":1592.9,"PV1Voltage":424.8,"PV1InputCurrent":2.2,"PV1InputPower":934.5,"PV2Voltage":411.5,"PV2InputCurrent":1.6,"PV2InputPower":658.4,"OutputPower":1736.8,"GridFrequency":49.91,"L1ThreePhaseGridVoltage":231.7,"L1ThreePhaseGridOutputCurrent":2.5,"L1ThreePhaseGridOutputPower":579.2,"L2ThreePhaseGridVoltage":231.5,"L2ThreePhaseGridOutputCurrent":2.4,"L2ThreePhaseGridOutputPower":555.6,"L3ThreePhaseGridVoltage":230.3,"L3ThreePhaseGridOutputCurrent":2.5,"L3ThreePhaseGridOutputPower":575.7,"TodayGenerateEnergy":7,"TotalGenerateEnergy":10801.8,"TWorkTimeTotal":2.0032424e7,"PV1EnergyToday":3.4,"PV1EnergyTotal":6544.9,"PV2EnergyToday":3.2,"PV2EnergyTotal":3779.5,"PVEnergyTotal":10324.4,"InverterTemperature":49,"TemperatureInsideIPM":35.3,"BoostTemperature":33.2,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"EnergyToUserTotal":0,"EnergyToGridToday":0,"EnergyToGridTotal":0,"DischargeEnergyToday":0,"DischargeEnergyTotal":0,"ChargeEnergyToday":0,"ChargeEnergyTotal":0,"LocalLoadEnergyToday":0,"LocalLoadEnergyTotal":0,"ACChargeEnergyToday":0,"ACChargeEnergyTotal":0,"Mac":"34:94:54:81:9F:FA","Cnt":364}

WaizenErnter, 2023-10-16:
I digged in to it a bit and fixed it. I only fixed the Battery SOC, Input and Output, because thats the only parameters i care about. My .bin file for the flasher is attached.
[firmware.zip](https://github.com/otti/Growatt_ShineWiFi-S/files/12921629/firmware.zip)

crasu, 2023-10-16:
I do not have a tl-x inverter with a hv battery. But if someone would like to test this #262 I could add support for this to master. You have to set the protocol to 6000.

WaizenErnter, 2023-10-17:
My version is already working.

crasu, 2023-10-18:
When someone wants to test #262 feel free to reopen

closed 2023-10-18 by crasu

newokatrix, 2024-02-08:
> I digged in to it a bit and fixed it. I only fixed the Battery SOC, Input and Output, because thats the only parameters i care about. My .bin file for the flasher is attached. [firmware.zip](https://github.com/otti/Growatt_ShineWiFi-S/files/12921629/firmware.zip)

@WaizenErnter : Could you please re upload the file ? The link is no longe working. Thank you !

Onlinepeter, 2024-08-16:
> My version is already working.

Have to tested beside the values you are able to receive if its also possible to send commands to the inverter eg change between load first, battery first or grid first?