OpenInverterGateway/OpenInverterGateway #249: Infos about GROWATT_MODBUS_VERSION for pre-comiled Releases

Reddingo, 2023-10-04:
# What happened?
From what I understood the supported MODBUS version is compiled into the binary.
To me it is not clear which GROWATT_MODBUS_VERSION was finally used for the pre-compiled versions under
https://github.com/otti/Growatt_ShineWiFi-S/releases

If possible, could you compile each version there with the 3-4 supported version?
so that you can have then 

- firmware-ShineWifiS-1.20.bin 
- firmware-ShineWifiS-1.25.bin 
- firmware-ShineWifiS-3.05.bin 
- firmware-ShineWifiX-1.20.bin 
- firmware-ShineWifiX-1.25.bin 
- firmware-ShineWifiX-3.05.bin 

(...)

# Stick type

all of them

# Which Inverter are you using?
- SPH-4000 BL-UP

I want to use the stick with my SPH-4000 and do not know yet which version I have to use (most propably 1.24).

Thank you

Reddingo, 2023-10-04:
Can you let me know which settings were used for the pre-compiled files that are currently uploaded under https://github.com/otti/Growatt_ShineWiFi-S/releases ?

crasu, 2023-10-04:
> Reference in ne

Sure it is here: https://github.com/otti/Growatt_ShineWiFi-S#how-to-install

Just updated the documentation.