OpenInverterGateway/OpenInverterGateway #250: More guidance needed for newbie to flashing

Vinisz, 2023-10-04:
Hi,

This looks good and I know about MQTT and all those kind of things, but....
That "flashing" (shinex) , how do I do that ? I only see 2 photos of the stick (it seems) with also some soldering ?
Do I need to insert the stick in my computer and flash it with a tool or something ?

I hope someone can share step-by-step instructions on these kind of things

DutchKillerbee, 2023-10-04:
Hello Vinisz,

Let me try to help you. This what I did in a few simple steps:

1. Download Arduino (from the Windows store)
2. Install the the ESP8266 board using the board manager under the tools menu
[image]
3. Install the required libraries using the library manager 
   - WiFiManager         by tzapu                  https://github.com/tzapu/WiFiManager
   - PubSubClient        by Nick O´Leary           https://github.com/knolleary/pubsubclient
   - DoubleResetDetector by Khai Hoang             https://github.com/khoih-prog/ESP_DoubleResetDetector
   - ModbusMaster        by Doc Walker             https://github.com/4-20ma/ModbusMaster
   - ArduinoJson         by Benoit Blanchon        https://github.com/bblanchon/ArduinoJson
   - Preferences         by vshymanskyy            https://github.com/vshymanskyy/Preferences
   - TeeLog              by Dirk-Willem van Gulik  https://github.com/dirkx/tee-log.git

5. Download the master source code as a sip file from Github
6. Extract the zip file into a folder
7. Rename the file config.sample.h to config.h and check the settings in config.h
8. Open the ShineWiFi-ModBus.ino file in Arduino
9. Compile and look for some errors (maybe some libraries are missing, Google is you friend....)
[image]

10. Now connect the GPIO0 pin hole with the ground pin hole on the board using a small "jumper wire".

11. Next connect your device to a USB port (keep the jumper wire in !). Two LED's (red and blue) will burn on your board 

12. Now hit the upload button (next to the compile button)

13. Your device will be flashed.......

14. When you receive a message "Hit hard RST button" your device should be ready and you need to remove it from your USB port.

15. Now remove the jumper wire and put the stick back in your PC.

16. If everything is OK the stick should go into AP mode and you should be able to configure it using your phone.

Sorry for my terrible English, I am a complete noob and using Arduino for the first time.......
I hope these steps will help you.

Br Martijn

crasu, 2023-10-04:
@DutchKillerbee Thank you for your efforts - but building the firmware with the arduino ide is not supported anymore use platform io.

@Vinisz You did not include any information on which stick type you are using. I guess its a wifi x so flashing documentation for this stick is here: 
https://github.com/otti/Growatt_ShineWiFi-S/blob/master/Doc/ShineWiFi-X/ShineWiFi-X.md#loading-firmware

There are various guide on the internet on how to flash an esp32.

You can even flash directly from the web to your usb connected esp:
https://esp.huhn.me/

P.S.
If some want to integrate something like this I would delighted:
https://github.com/witnessmenow/ESP-Web-Tools-Tutorial

closed 2023-10-04 by crasu