OpenInverterGateway/OpenInverterGateway #251: Error: Unknown Shine Stick

mphquer, 2023-10-04:
Hi,
i created a new version and loaded it onto the shine x stick (version 2022-11-01).
Now I get the following entry in the log file:

// Short sequence from log file
Setup() Opened serial web server on http://(IP unset):8080 
WIFI connected...yeey :) 
MqttServer: 192.168.1.20 
MqttPort: 1883 
MqttTopic: energy/pv 
Error: Unknown Shine Stick

Why is the error generated?

Juergen

DutchKillerbee, 2023-10-04:
Hi Juergen,
Call me stupid but did you insert the stick in the Growatt Inverter?
DKB

crasu, 2023-10-04:
@mphquer To answer your question i need the following information

What Stick type are you using?
- Shine X
- Shine S
- Lolin32
- Nodemcu32
- Other 

Which Inverter are you using?
- MIC-1000-TL-X
- SPH-3000 BL-UP
- ....

How is the stick attached to the inverter?

mphquer, 2023-10-04:
> Hi Juergen, Call me stupid but did you insert the stick in the Growatt Inverter and set SIMULATE_INVERTER to 0? DKB

Hi DBK,

yes, i but the stick in the inverter.
#config.h
#define SIMULATE_INVERTER 0
#define SIMULATE_DEVICE ShineWiFi_X

Jürgen

mphquer, 2023-10-04:
> @mphquer To answer your question i need the following information
>
> What Stick type are you using?
>
>     * Shine X
>
>     * Shine S
>
>     * Lolin32
>
>     * Nodemcu32
>
>     * Other
>
>
> Which Inverter are you [...]

The stick is an shine x.
I use an MIC 600TL-X inverter.
The stick is connected directly to the inverter via USB.

Jürgen

mphquer, 2023-10-05:
Hi,

the problem is solved. Of course, the error message only came when the stick was plugged into the notebook.
My mistake, sorry.

Jürgen

closed 2023-10-06 by crasu