OpenInverterGateway/OpenInverterGateway #255: Mqtt sensor state is not retained after restart HA

DutchKillerbee, 2023-10-05:
# What happened?

First of all I's like to thank you for this firmware!
Everything works perfect for me but I have one annoying issue I cannot solve.

Everytime when I restart HA (when my inverters are off) the sensor state of my HA sensors become "unavailable" and the sensor value of the TotalGeneratedEnergy becomes 0. 
[image]
I have two growatt inverters and two flashed ShineWIFI-X sticks. Both sticks publish in seperate topics
[image]

My sensor settings of both mqtt sensors:
```
  sensor:
    - unique_id: "growatt_garage_total_production"
      state_topic: "growatt/garage"
      name: "Growatt.Garage.TotalGenerateEnergy"
      unit_of_measurement: "kWh"
      value_template: "{{ float(value_json.TotalGenerateEnergy) | round(1) }}"
      device_class: energy
      state_class: total_increasing
      json_attributes_topic: "growatt/garage"
      payload_available: "1"
      availability_mode: latest
      availability_topic: "growatt/garage"
      availability_template: "{{ value_json.InverterStatus }}"
    - unique_id: "growatt_veranda_total_production"
      state_topic: "growatt/veranda"
      name: "Growatt.Veranda.TotalGenerateEnergy"
      unit_of_measurement: "kWh"
      value_template: "{{ float(value_json.TotalGenerateEnergy) | round(1) }}"
      device_class: energy
      state_class: total_increasing
      json_attributes_topic: "growatt/veranda"
      payload_available: "1"
      availability_mode: latest
      availability_topic: "growatt/veranda"
      availability_template: "{{ value_json.InverterStatus }}"

```
Do you have a suggestion how I can keep the value as soon as I restart HA or even when I restart the server?

Thanks

DutchKillerbee, 2023-10-06:
Some addional information.
I am using mosquitto 2.0.11 so this potential issue is not applicable.
[https://github.com/eclipse/mosquitto/issues/2887](url)

crasu, 2023-10-07:
I think this is more a modbus issue then an issue with the firmware. Afaik sensor data in HA is transient data. For durable statistic I use the energy dashboard (https://www.home-assistant.io/dashboards/energy/).

closed 2023-10-07 by crasu