OpenInverterGateway/OpenInverterGateway #259: How can i change the Protocoll to 120?

WaizenErnter, 2023-10-15:
Hello guys,

Im allready using this firmware on my Shine Wifi X. But i cant read the SOC or the other Battery values. It looks like the prebuild .bin is fixed on Protocoll 124 but i need 120 for my inverter. Sorry for the question, im new into this.

Kochi1316, 2023-10-15:
Hi,
you need to download the project and compile it yourself.
In the Config.h.example file, you can customize the Modbus version according to your specific requirements.
Afterward, save the file, rename it to Config.h, and compile the project.

WaizenErnter, 2023-10-15:
How do I compile it into an .bin file?

Kochi1316, 2023-10-15:
You can do that using PlatformIO.
There are many tutorials on how to install Visual Studio Code and the PlatformIO plugin, allowing you to compile projects with it.

closed 2023-10-16 by crasu