OpenInverterGateway/OpenInverterGateway #260: Write Modbus HoldigRegister via HTTP POST

Kochi1316, 2023-10-15:
# What happened?
Laut diesem Post von Bernd9000 https://github.com/otti/Growatt_ShineWiFi-S/issues/20#issuecomment-1259645121 ist es möglich die Holding Register mittels HTTP POST zu ändern um die Leistung des Wechselrichters zu ändern. Ich habe es als erstes mit Node RED versucht. Als ich immer wieder ein "Invalid Request" bekam habe ich es mit Postman auf Windows versucht. Auch dort bekam ich einen "Invalid Request".
Das JSON-Objekt sieht folgendermaßen aus:

**in english**
According to this post by Bernd9000 on https://github.com/otti/Growatt_ShineWiFi-S/issues/20#issuecomment-1259645121, it is possible to change the holding registers using HTTP POST to modify the inverter's performance. I initially attempted this using Node RED. When I repeatedly received an "Invalid Request" error, I tried it with Postman on Windows. I encountered the same "Invalid Request" issue there as well. The JSON object looks as follows:
`{
    "reg": 3,
    "val": 42,
    "type": "16b",
    "operation": "W",
    "registerType": "H"
}`

Der Node RED Flow sieht folgendermaßen aus:

**in english**
The Node RED flow looks as follows:

`[
    {
        "id": "05dc5e4e1f072888",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "b393e8307dd31994",
        "type": "debug",
        "z": "05dc5e4e1f072888",
        "name": "debug 7",
        "active": true,
        "tosidebar": true,
        "console": true,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 750,
        "y": 280,
        "wires": []
    },
    {
        "id": "a34bdba9a6c1a3b8",
        "type": "http request",
        "z": "05dc5e4e1f072888",
        "name": "",
        "method": "POST",
        "ret": "txt",
        "paytoqs": "body",
        "url": "http://192.168.0.80/postCommunicationModbus_p",
        "tls": "",
        "persist": false,
        "proxy": "",
        "insecureHTTPParser": false,
        "authType": "",
        "senderr": false,
        "headers": [],
        "x": 750,
        "y": 240,
        "wires": [
            [
                "b393e8307dd31994"
            ]
        ]
    },
    {
        "id": "d55126296b26533c",
        "type": "change",
        "z": "05dc5e4e1f072888",
        "name": "",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "{\t    \"reg\": 3,\t    \"val\": msg.payload,\t    \"type\": \"16b\",\t    \"operation\": \"W\",\t    \"registerType\": \"H\"\t}",
                "tot": "jsonata"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 530,
        "y": 240,
        "wires": [
            [
                "a34bdba9a6c1a3b8",
                "b393e8307dd31994"
            ]
        ]
    },
    {
        "id": "e8606b25bb58ca8e",
        "type": "inject",
        "z": "05dc5e4e1f072888",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "65",
        "payloadType": "num",
        "x": 310,
        "y": 240,
        "wires": [
            [
                "d55126296b26533c"
            ]
        ]
    }
]`

Benötigt der Webserver auf dem esp ein anderes Format?

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MIC 600TL-X

# Log output
Please copy and paste any relevant log output from /debug and the serial console (You have to compile the firmware with the relevant settings in Config.h to get the output).

```
400 Bad Request
```

Für Hinweise wäre ich sehr dankbar.

crasu, 2023-10-17:
The route will not accept json input as it is not a rest service.

Here is a curl example on how to set the output power for the inverter:
```
curl 'http://<ip>/postCommunicationModbus_p' \
  --data-raw 'reg=3&val=25&type=16b&operation=W&registerType=H' \                                                                                       
  --compressed 
Wrote Holding Register 3 to a value of 25!
```

closed 2023-10-17 by crasu

SimonBard, 2024-03-25:
I wanted to increase the max AC output from 600 to 2000W, but I get an error: 

```
$ curl 'http://192.168.178.200/postCommunicationModbus_p' --data-raw 'reg=121&val=20&type=16b&operation=W&registerType=H' --compressed
Read 16b Holding register 121 impossible - not connected?
```
I have connection via browser and values are constantly updated. 
I get the same error when I try to write the register via: 
[](http://192.168.178.200/postCommunicationModbus)

[image]

Kochi1316, 2024-04-07:
Hi Simon,

Sorry for the late reply.
It's not possible to provide performance values that exceed those of the inverter. That's why the error occurs.
The maximum power of the inverter is directly embedded in the firmware and cannot be changed through the ModBUS registers.