OpenInverterGateway/OpenInverterGateway #267: energy/solar/command/batteryfirst/get doesn't return a result

i-am-shodan, 2023-10-27:
# What happened?
Run this command:
mosquitto_pub -m "{\\"correlationId\\": \\"ha-batteryfirst-get\\"}" -t energy/solar/command/batteryfirst/get

Expecting to get a response from in a similar way to the get/set time example
mosquitto_sub.sh -t energy/solar/result

Never get a response, stick seems up ie it hasn't crashed

# Stick type
- Shine S

Which Inverter are you using?
- SPH-5000 BL-UP

# Log output
No logs /debug hangs but I do get other results over JSON and setting the time work as per the example

i-am-shodan, 2023-10-28:
I've now tried with the empty json payload referenced in the PR https://github.com/otti/Growatt_ShineWiFi-S/pull/210

Same issue

i-am-shodan, 2023-10-28:
@MattJeanes you able to shed any light on why no result might be returned? I've got getting and setting time working.

MattJeanes, 2023-10-30:
Hey @i-am-shodan managed to miss a few emails sorry about the wait

You should get something in your `/debug` log every time you run any sort of command, even an invalid one

In this case, you should see `Handling command: batteryfirst/get`

This is what my log looks like when I call it a few times:

[image]

I am running a slightly older firmware though as there's been a few commits since I last updated mine

I wonder if you see any different on the same commit I am using? https://github.com/otti/Growatt_ShineWiFi-S/commit/0de73d81efec281c51da734f9e8e99746e73c629

There's been a few changes in related areas that could potentially break things https://github.com/otti/Growatt_ShineWiFi-S/compare/0de73d81efec281c51da734f9e8e99746e73c629...master

i-am-shodan, 2023-10-30:
I'm getting

```
.. skipping 2927100 bytes of log - no longer in buffer ..
age... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
MQTT message arrived [energy/solar/command/batteryfirst/get] { "correlationId": "get-timeslot" } 0 14:01:03","command":"datetime/get","success":true,"message":"Successfully read date/time"}
Handling command: batteryfirst/get
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
ReadData() successful
publish MQTT message... succeed
```

i-am-shodan, 2023-10-30:
I don't like the `"Successfully read date/time"` is the wrong command being executed?

i-am-shodan, 2023-10-30:
Ok i've added a load of new debugging and reflashed. I think it'd running the function but not writing the result to the JSON. Will take a more detailed look later.

MattJeanes, 2023-10-30:
I recommend trying the older commit I mentioned too if you haven't already, I am interested to see if this is a regression in newer firmware

Let me know what you find, I'm interested to see how this turns out!

i-am-shodan, 2023-10-30:
I run a fork of this code base so rolling forward is the only option for me. I think I understand the problem, just need to sit down with the code.

i-am-shodan, 2023-11-01:
I've added a lot more debug to the ShineMqtt::onMqttMessage function
[image]

Results are confusing. It is creating the right payload, it says it sends it off but I never get it on the other end - unlike the date time messages
[image]

I'm going to try and make the message smaller and see what happens

i-am-shodan, 2023-11-01:
I've commented out this code to make the json smaller

[image]

I now get a response

i-am-shodan, 2023-11-01:
I've rolled back a lot of changes. It looks like maybe the message is too big for the device send maybe? The message looks like it's formed right and mqttclient.publish returns true but I don't actually receive anything on my mosquitto server until the message size is much smaller.

I'm going to now try to only return 1 slot.

i-am-shodan, 2023-11-01:
What I don't get is that the usual inverter status messages are much bigger anyway. Maybe it's something to do with nested object serialisation?

i-am-shodan, 2023-11-01:
Worth saying in some debug out i've seen fragments of HTTP requests in the JSON. I was pulling the logs out via curl and I had assumed this was a result of that. Now I'm thinking that what actually is happening is some memory corruption somewhere.

crasu, 2023-11-02:
IMHO the code at the end of handle creates at least 2 copies of the the entire output json on the stack. If someone is willing to test this I can convert this two to the arduino json streaming api and use the global static json buffer (which would reduce memory consumption by quite a bit). @i-am-shodan can you figure out how big the resulting json is (by calling measureJson())?

closed 2023-11-15 by crasu

MattJeanes, 2023-11-15:
@crasu I have tried this latest commit and it is now only returning empty JSON responses containing only the correlation ID if set on the request. The logs appear normal and as far as I can tell appears to be processing requests normally just no data out of the result end.

I've reverted to https://github.com/otti/Growatt_ShineWiFi-S/commit/3a2002e6062c5f3c2e6402dad5281024ed3ba633 for the mean time but let me know if I can help test any fixes to this.

Side note: The firmware update being only on the config AP on newer commits is a bit annoying, would still be nice to have the option to do it from the main UI as I can only get into the config AP if I stand really close to the inverter so I have to load the bin file onto my phone, go outside to the inverter in the shed and then connect to the access point and upload the firmware, was much easier previously when I could upload a new one sat at my desk over LAN

i-am-shodan, 2023-11-16:
Agreed that flashing is now a PITA

commit 93b94a0e (2023-11-15): Stream mqtt json output to socket in order to avoid buffer overruns. Fixes #267