OpenInverterGateway/OpenInverterGateway #268: It would be great to get an example of setting battery first mode with a timespan with MQTT. Looking at the code it looks like a lot of work has gone into setting the right registers but without a good example it's difficult to use this feature.

i-am-shodan, 2023-10-28:


i-am-shodan, 2023-10-28:
It would be great to get an example of setting battery first mode with a timespan with MQTT. Looking at the code it looks like a lot of work has gone into setting the right registers but without a good example it's difficult to use this feature.

MattJeanes, 2023-10-30:
Here's an example request to set slot 1 of the 3 timeslots to charge between 23:30 - 05:30:

`energy/solar/command/batteryfirst/set/timeslot`
```json
{
    "correlationId": "set-timeslot",
    "slot": 1,
    "start": "23:30",
    "stop": "05:30",
    "enabled": true
} 
```

Response on `energy/solar/result`
```json
{
    "correlationId": "set-timeslot",
    "command": "batteryfirst/set/timeslot",
    "success": true,
    "message": "success"
}
```

And here is another just to get all the information including all timeslots:

`energy/solar/command/batteryfirst/get`
```json
{
    "correlationId": "get-timeslot"
} 
```

Response on `energy/solar/result`
```json
{
    "correlationId": "get-timeslot",
    "powerRate": 100,
    "stopSOC": 100,
    "acChargeEnabled": true,
    "timeSlots": [
        {
            "slot": 1,
            "start": "23:30",
            "stop": "05:30",
            "enabled": true
        },
        {
            "slot": 2,
            "start": "00:00",
            "stop": "00:00",
            "enabled": false
        },
        {
            "slot": 3,
            "start": "00:00",
            "stop": "00:00",
            "enabled": false
        }
    ],
    "command": "batteryfirst/get",
    "success": true,
    "message": "success"
}
```

Hope that helps, might be worth putting in a PR to edit the docs if you find that helpful!

Another thing that could be helpful is that I've converted the command API into a REST API using C# ASP.NET Core for my own use: https://github.com/MattJeanes/HomeAutomation/blob/main/HomeAutomation.Web/Controllers/EnergyController.cs

i-am-shodan, 2023-10-30:
Thanks, I've tested a few more mqtt topics. Using your example for energy/solar/command/batteryfirst/set/timeslot I can get a response in energy/solar/result. But for energy/solar/command/batteryfirst/get I get nothing.

SebastianGode, 2023-11-15:
@MattJeanes 
How can I enable or disable the `acChargeEnabled` toggle?
I guess I could program the timeslot i want once and then just toggle this value, right?

i-am-shodan, 2023-11-15:
I'm using this automation. I uses some home assistant input helpers which I have on a dashboard to set the right values.

```

alias: Growatt - Set timeslot
description: ""
trigger:
  - platform: event
    event_type: ""
condition: []
action:
  - service: mqtt.publish
    data:
      qos: "1"
      topic: energy/solar/command/batteryfirst/set/timeslot
      payload_template: |
        { 
          "correlationId": "set-timeslot", 
          "slot": 1, 
          "start": "{{ states('input_datetime.input_datetime_growatt_batteryfirst_slot1_start_time')[:5] }}", 
          "stop": "{{ states('input_datetime.input_datetime_growatt_batteryfirst_slot1_stop_time')[:5] }}", 
          "enabled": {{ is_state('input_boolean.growatt_batteryfirst_slot1_enabled', 'on') | lower }}
        }
  - service: mqtt.publish
    data:
      qos: "1"
      topic: energy/solar/command/batteryfirst/set/acchargeenabled
      payload_template: |
        { 
          "correlationId": "set-acchargeenabled", 
          "value": {{ is_state('input_boolean.growatt_batteryfirst_accharge_enabled', 'on') | lower }}
        }
mode: single

```

i-am-shodan, 2023-11-15:
Also worth saying the inverter I'm using keeps terrible time. You may need to set the time every day to avoid terrible drift.

MattJeanes, 2023-11-15:
@i-am-shodan there is a separate command to set `acChargeEnabled` which is `batteryfirst/set/acchargeenabled` as you have seen yes, it is set independently of any of the timeslots and applies to all of them.

You can get a full list of commands by using the `list` command:
```json
{
    "commands": [
        "batteryfirst/get",
        "batteryfirst/set/acchargeenabled",
        "batteryfirst/set/powerrate",
        "batteryfirst/set/stopsoc",
        "batteryfirst/set/timeslot",
        "datetime/get",
        "datetime/set",
        "echo",
        "gridfirst/get",
        "gridfirst/set/powerrate",
        "gridfirst/set/stopsoc",
        "gridfirst/set/timeslot",
        "list",
        "modbus/get",
        "modbus/set"
    ],
    "command": "list",
    "success": true,
    "message": ""
}
```

I have also noticed the inverter keeps terrible time, and I have a HA automation set up to update this frequently for the same reason.

SebastianGode, 2023-11-15:
> I have also noticed the inverter keeps terrible time, and I have a HA automation set up to update this frequently for the same reason.

Can you please post the automation here quickly?

Also as I never added automations manually through YAML, where can I do this exactly? Also just a new YAML file somewhere?

i-am-shodan, 2023-11-15:
acchargeenabled is the second command in my automation above.

The first MQTT publish command sets the time, the second ac enabled.

i-am-shodan, 2023-11-15:
> Can you please post the automation here quickly?
>
>
>
> Also as I never added automations manually through YAML, where can I do this exactly? Also just a new YAML file somewhere?

It's above. Create a new automation in the UI, find the button to edit in yaml mode and paste the above it. It relies on some input helpers. You can creating them in the UI as well.

MattJeanes, 2023-11-15:
This is my automation for keeping the inverter time up to date:

```yaml
alias: Update inverter date/time
description: ""
trigger:
  - platform: time_pattern
    enabled: true
    hours: "*"
condition: []
action:
  - service: mqtt.publish
    data:
      qos: "1"
      topic: energy/solar/command/datetime/set
      payload_template: |
        {
          "correlationId": "ha-datetime-set",
          "value": "{{ now().strftime('%Y-%m-%d %H:%M:%S') }}"
        }
mode: single
```

As @i-am-shodan mentions, add a new automation and switch to YAML mode with the 3 dots at the top right

SebastianGode, 2023-11-15:
Thanks a lot!

Just a quick questions still:
You provided the template 
```
        {
          "correlationId": "ha-datetime-set",
          "value": "{{ now().strftime('%Y-%m-%d %H:%M:%S') }}"
        }
```
What does `correlationId` do and what's "ha-datetime-set"?

MattJeanes, 2023-11-15:
Sure, the `correlationId` is just a value you send with a command and it gets sent back with the result

This is so you can uniquely identify and associate your request to the response if there are multiple commands being sent.

In my case, I use this along with one for `ha-datetime-set` to associate them to HA entities which I can then run further automations on, for example if the command fails to execute.

```yaml
- trigger:
    platform: mqtt
    topic: energy/solar/result
    value_template: "{{ value_json.correlationId }}"
    payload: "ha-datetime-set"
  binary_sensor:
    name: Growatt - Inverter set date/time
    state: "{{ trigger.payload_json.success }}"
    attributes:
      message: "{{ trigger.payload_json.message }}"
```

I also have another automatic that call's the datetime get with `ha-datetime-get` to get that information as well

```yaml
- trigger:
    platform: mqtt
    topic: energy/solar/result
    value_template: "{{ value_json.correlationId }}"
    payload: "ha-datetime-get"
  sensor:
    name: Growatt - Inverter date/time
    state: "{{ trigger.payload_json.value }}"
    attributes:
      success: "{{ trigger.payload_json.success }}"
      message: "{{ trigger.payload_json.message }}"
```

And the associated automation:

```yaml
alias: Get inverter date/time
description: ""
trigger:
  - platform: time_pattern
    minutes: "*"
condition: []
action:
  - service: mqtt.publish
    data:
      qos: "1"
      topic: energy/solar/command/datetime/get
      payload: "{\"correlationId\": \"ha-datetime-get\"}"
mode: single
```

SebastianGode, 2023-11-15:
Omg, HomeAssistant can do way way more than I thought.
Crazy, really need to read into the docs for all that stuff at some point.

SebastianGode, 2023-11-17:
I just tried the following:
```yaml
alias: Get Inverter BatteryFirst Status
description: ""
trigger:
  - platform: event
    event_type: ""
condition: []
action:
  - service: mqtt.publish
    data:
      qos: "1"
      topic: energy/solar/command/batteryfirst/get
      payload_template: |
        {
            "correlationId": "get-timeslot"
        } 
mode: single

```
This however will result in the following output in `energy/solar/result`:
```json
{
  "correlationId": "get-timeslot",
  "command": "batteryfirst/get",
  "success": false,
  "message": "Failed to read battery first settings"
}
```

What's the issue here?

i-am-shodan, 2023-11-17:
You can try to flash the codebase in my repo, that works.

MattJeanes, 2023-11-17:
The configuration looks good to me, I ran exactly the same automation and it is returning results on my end:

```json
{
  "correlationId": "get-timeslot",
  "powerRate": 100,
  "stopSOC": 100,
  "acChargeEnabled": true,
  "timeSlots": [
    {
      "slot": 1,
      "start": "23:30",
      "stop": "05:30",
      "enabled": true
    },
    {
      "slot": 2,
      "start": "00:00",
      "stop": "00:00",
      "enabled": false
    },
    {
      "slot": 3,
      "start": "00:00",
      "stop": "00:00",
      "enabled": false
    }
  ],
  "command": "batteryfirst/get",
  "success": true,
  "message": "success"
}
```

I am using a slightly older commit of this firmware though: https://github.com/otti/Growatt_ShineWiFi-S/commit/3a2002e6062c5f3c2e6402dad5281024ed3ba633

crasu, 2023-11-17:
The issue with the code base is that it is using a large strings on the heap. Which is causing heap fragmentation and crashes the stick after some time. I reduced the buffer size a couple of month ago this seems to have broken the SPH code.

crasu, 2023-11-17:
@SebastianGode you can try this pr #279. The fix was super simple - I just forgot to push one commit.

SebastianGode, 2023-11-18:
I will try today I guess. Excited to see it working :)

Off topic:
I have installed 10kWh ARK HV batteries by my electrician and then bought 2 additional ones immediately afterwards and connected them. Voltage seems fine and is higher with the new packs, but I somehow only get around 7kWh still from the whole packs under full load. This makes sense for the 10kWh version, but not for 15kWh now. Do I need to reconfigure the inverter or something like that? Or did it need a first complete charging cycle to learn the new capacity? Completely off topic here, I know, but maybe someone has an idea ^^

crasu, 2023-11-21:
@SebastianGode Did you get around to test this?

SebastianGode, 2023-11-21:
@crasu 
I sadly have an ticket for my battery issue open with Growatt and they need access through their ShineWifi App.
So I need to use the original firmware in the meantime (grrrrr)

MattJeanes, 2023-11-21:
I'm happy to test this but I ideally need the ability to flash firmware from the web UI back in again cause it's a royal pain to do it with the config AP for me

crasu, 2023-11-21:
@MattJeanes I removed it to save some memory and because its a security issue on the precompiled fw. Because everbody can get the credentials from the repo. I will have a look into it in the coming days.

MattJeanes, 2023-11-21:
Yeah makes sense to have it off by default due to the precompiled FW as you say but having an option to turn it on would be fantastic, thank you!

crasu, 2023-12-08:
Documentation issue fixed by #274

crasu, 2023-12-08:
Add #286 for the update issue

closed 2023-12-08 by crasu

InvertedMinor, 2025-05-02:
I am a bit puzzled, in my case I got the following command list:

{
  "commands": [
    "bdc/set/acchargeenabled",
    "bdc/set/chargepowerrate",
    "bdc/set/chargestopsoc",
    "bdc/set/dischargepowerrate",
    "bdc/set/dischargestopsoc",
    "datetime/get",
    "datetime/set",
    "echo",
    "list",
    "modbus/get",
    "modbus/set",
    "power/get/activerate",
    "power/set/activerate",
    "priority/set"
  ],
  "command": "list",
  "success": true,
  "message": ""
}

What I am doing wrong?