OpenInverterGateway/OpenInverterGateway #269: ShineWifiX no more Flash possible after Image 1.0

Stefan333de, 2023-11-03:
Hello,

have an ShineWifiX flashed some months ago with Image 1.0 with flash_download_tool_3.9.4
All ok, works fine.
Meanwhile i bought a second ShineX to learn to flash with IDE, but after i see the new Image 2.01 i flashed this factory Stick with flash download tool also...:)

No Problems, 2.01 works now in the MIN4600.
Today i tried to update the old Stick to new Release but it will not work. 

Same computer and settings as yesterday with the new factory Stick.

But I get only this Error:
-------------------------------------------------
test offset :  0 0x0
case ok

[2023-11-03 20:45:30,334][ESP8266Loader_spi[1]][espDownloader.py][line:566][ERROR]: IOError: the serial port should probably be removed
[2023-11-03 20:45:30,334][ESP8266Loader_spi[1]][espDownloader.py][line:759][ERROR]: ESP8266 Chip sync error esp_sync_blocking.
no log file output ...
---------------------------------------------------
Also erase Flash is not possible.
Com Port and Com Speed double checked, setting is both 115200

Is the Stick bricked?

regards
Stefan

crasu, 2023-11-04:
I have never used the flash_download_tool - So I am unable to assist. I doubt the issue is caused by the firmware itself. Please double check your cabling and the setup. It can be a bit tricky at time. I use esptool.

closed 2023-11-04 by crasu

Stefan333de, 2023-11-04:
Well, this super reliable Win10 is the culprit....:(
Went to another maschine with the Software Tools and all works fine. Checked the difference between them and on the first PC there was no more Exar Com driver 

In the 24h between the two flash operation it has illegitimate changed the com Port driver to Microsoft driver.

regards
Stefan