OpenInverterGateway/OpenInverterGateway #272: Cant write to ShineWifi-X

Schelin78, 2023-11-08:
# What happened? Cant write to ShineWifi-X
# Stick type - Shine X
# Which Inverter are you using? Growatt Max 50KTL3 LV
# Log output

In file included from C:\Users\KBV\Downloads\Growatt_ShineWiFi-S-master (1)\Growatt_ShineWiFi-S-master\SRC\ShineWiFi-ModBus\ShineWiFi-ModBus.ino:27:
C:\Users\KBV\Downloads\Growatt_ShineWiFi-S-master (1)\Growatt_ShineWiFi-S-master\SRC\ShineWiFi-ModBus\Config.h:48:1: warning: multi-line comment [-Wcomment]
   48 | //#define MQTTS_BROKER_CA_CERT "-----BEGIN CERTIFICATE-----\n" \
      | ^
. Variables and constants in RAM (global, static), used 41540 / 80192 bytes (51%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1528     initialized variables
╠══ RODATA   6516     constants       
╚══ BSS      33496    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 60891 / 65536 bytes (92%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28123    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 420008 / 1048576 bytes (40%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     420008   code in flash   
esptool.py v3.0
Serial port COM3
Connecting......

A fatal esptool.py error occurred: Write timeout

* Have put a paper clip between GP100 and GND. Then blue and red leds light up, Green is not lit. Tried various things, left it on all the time, started it with clip and then removed it, same problem.
[image]
[image]

* Downloaded https://github.com/otti/Growatt_ShineWiFi-S/tree/master/SRC/ShineWiFi-ModBus
* Downloaded and started Arduino IDE 2.2.1.
* Opened ShineWiFi-ModBus.ino
* Selected board "Generic ESP8266 Module" on COM3
* Getting this error message "A fatal esptool.py error occurred: Write timeout"
* Am I missing a driver? What kind of driver should it be for Shinewifi-X? Or what am I doing wrong?

Schelin78, 2023-11-15:
Thanks but I didn't find the answer there?

Schelin78, 2023-11-15:
Still no solution

SvenPook, 2023-11-15:
I've encountered the same. Seemed like a lot of packages were missing using the Arduino IDE. I can recommend using PLATFORMIO in Visual Studio Code. Flashing the firmware was a walk in the park using that.

closed 2023-11-15 by crasu

ViRuSTriNiTy, 2024-05-26:
You get the write timeout error because your stick most likely has a different USB chipset, therefore you need a different USB driver to make it work. My sticks come with `XR21V1410`. The drivers can be found here: https://www.maxlinear.com/product/interface/uarts/usb-uarts/xr21v1410

Basically, first try to identify the chips on your stick using a magnifier.