OpenInverterGateway/OpenInverterGateway #278: SPH-60000TL3-BH-UP - "Unknown Shine Stick"

MarcoKu, 2023-11-16:
# What happened?
I have a SPH-60000TL3-BH-UP. I bought a second Shine X stick and flashed firmware-ShineWifiX.bin.
I replaced the original stick with the one new flashed and configured everything in the webclient. 
Afertwards the led is blinking green and no values are comming from MODBUS. Writing MQTT is working.
Then I loaded the source, configured everything (Protocoll 124) and uploaded it with Arduino IDE.
The same output. I do not get connected the the Modbus from Inverter.

# Stick type
- Shine X

Which Inverter are you using?
- SPH-60000TL3-BH-UP

# Log output
...
GetWiFiStickType: 
Error: Unknown Shine Stick
...

Can someone pleas help me! What I have to do to get Modbus working?

Thank you and best regards!
Marco

remco-h, 2023-11-18:
Same problem here. I have a Growatt MIC 1500 TL-X with the ShineWiFi-X stick. The stick was working before i flashed the firmware-ShineWifiX.bin file.

The data values stays 0. 

In de log i get the messages: Unknown Shine Stick

Setup()
Opened serial web server on http://(IP unset):8080
WIFI connected...yeey :)
MqttServer: 192.168.10.10
MqttPort: 1883
MqttTopic: energy/solar
Error: Unknown Shine Stick
MqttServer: 192.168.10.10
MqttUser: Mqtt
MqttTopic: energy/solar
Attempting MQTT connection...connected
Subscribed to energy/solar/command/#
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick

MarcoKu, 2023-11-20:
Hi, I could solve my problem with changing the Slave-ID to 4. 
But you need to compile it by yourself to do this changing.

Best regards,
Marco

remco-h, 2023-11-21:
The stick is working with changing the Slave-ID.  i had another remark with the Version for protocol 3.05 in HA. The data values that the MQTT recieved are diffrent. in the list below the names and how i soleved it to get the right data in HA. 
`####################### Growatt solar panel converter MQTT #################################

mqtt:
  sensor:
    - name: "Growatt DC Voltage"
      unique_id: "growatt_dc_voltage"
      unit_of_measurement: "V"
      device_class: "voltage"
      state_topic: "energy/solar"
      value_template: "{{ value_json.PV1Voltage }}"
      
    - name: "Growatt DC Current"
      unique_id: "growatt_dc_current"
      unit_of_measurement: "A"
      device_class: "current"
      state_topic: "energy/solar"
      value_template: "{{ value_json.PV1InputCurrent }}"

    - name: "Growatt DC Input Power"
      unique_id: "growatt_dc_input_power"
      unit_of_measurement: "W"
      device_class: "current"
      state_topic: "energy/solar"
      value_template: "{{ value_json.InputPower }}"
      
    - name: "Growatt AC Frequency"
      unique_id: "growatt_ac_frequency"
      unit_of_measurement: "Hz"
      state_class: "measurement"
      state_topic: "energy/solar"
      device_class: "frequency"
      value_template: "{{ value_json.GridFrequency }}"  

    - name: "Growatt AC Voltage"
      unique_id: "growatt_ac_voltage"
      unit_of_measurement: "V"
      state_class: "measurement"
      device_class: "voltage"
      state_topic: "energy/solar"
      value_template: "{{ value_json.L1ThreePhaseGridVoltage }}" 

    - name: "Growatt AC Current"
      unique_id: "growatt_ac_Current"
      unit_of_measurement: "A"
      state_class: "measurement"
      device_class: "current"
      state_topic: "energy/solar"
      value_template: "{{ value_json.L1ThreePhaseGridOutputCurrent }}"       
  
    - name: "Growatt AC Power"
      unique_id: "growatt_ac_power"
      unit_of_measurement: "W"
      state_class: "measurement"
      device_class: "power"
      state_topic: "energy/solar"
      value_template: "{{ value_json.L1ThreePhaseGridOutputPower }}"              
      
    - name: "Growatt Temperature"
      unique_id: "growatt_temperature"
      unit_of_measurement: "°C"
      state_class: "measurement"
      device_class: "temperature"
      state_topic: "energy/solar"
      value_template: "{{ value_json.InverterTemperature }}"
      
    - name: "Growatt Energy Today"
      unique_id: "growatt_energy_today"
      unit_of_measurement: "kWh"
      state_class: "total_increasing"
      device_class: "energy"
      state_topic: "energy/solar"
      value_template: "{{ value_json.TodayGenerateEnergy }}" 
      
    - name: "Growatt. Energy Total"
      unique_id: "growatt_total_production"
      state_topic: "energy/solar"
      unit_of_measurement: "kWh"
      value_template: "{{ float(value_json.PV1EnergyTotal) | round(1) }}"
      device_class: energy
      state_class: total_increasing
      json_attributes_topic: "energy/solar"
      payload_available: "1"
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{ value_json.InverterStatus }}"

    - name: "Growatt Worktime"
      unique_id: "growatt_worktime"
      unit_of_measurement: "pending_time"
      state_topic: "energy/solar"
      value_template: "{{ value_json.TWorkTimeTotal }}"`

[energy_solar_growatt MIC 1500 TL-X.txt](https://github.com/OpenInverterGateway/OpenInverterGateway/files/13429520/energy_solar_growatt.MIC.1500.TL-X.txt)

MarcoKu, 2023-11-21:
Solution: Change Slave-ID

closed 2023-11-21 by MarcoKu

Neoster, 2024-05-05:
Hi @MarcoKu,

Thanks for documenting the process and solution.
I find myself stuck at the same error while trying to get the Growatt data into home assistant.
It seems I am stuck at exactly the same place, the modes does not seem to start up correctly, or even at all.

`Opened serial web server on http://(IP unset):8080
Setting up WiFiManager menu
WIFI connected...yeey :)
MqttServer: 192.168.2.250
MqttPort: 1883
MqttTopic: energy/solar
Modbus: init with slave id: 3
Modbus: init response: 0xE2
Modbus error: Response Timed Out
Error: Unknown Shine Stick
MqttServer: 192.168.2.250
MqttUser: homeassistant
MqttTopic: energy/solar
Attempting MQTT connection...connected
Subscribed to energy/solar/command/#`

I own a Growatt MIN 2500 TL-X inverter.
I tried flashing the ShineWifi-X module as well as a brand new NodeMCU V2, both resulting in the same 'Unknown Shine Stick'/ 'Response Timed Out' situation.

I also tried slave id 1, 3 and 4 for the modbus, no difference in outcome.
Is there a possibility you've got another suggestion for me to try?
Final question, how to i know which modbus version i should compile the code with?

Thanks for your time!

crasu, 2024-05-05:
@Neoster you probably have the wrong serial to usb converter on the nodemcu. Check the main readme for compatible boards.