OpenInverterGateway/OpenInverterGateway #284: PV energy at night / home assistant

truwer, 2023-11-26:
Hello, thank you very much for this awesome project. It works perfectly expect one small thing. Not sure if this is an configuration issue or if don't understand the reason.

# What happened?
The stick sends minor energy production at night. I can't imagine that I produce PV energy at night. The battery does not discharge and in parallel there is not PV input or outout.

[image]
[image]
[image]

# Stick type
What Stick type are you using?
- Shine X

Which Inverter are you using?
SPH4~10k TL3 BH

# Log output
If you want to see any logs, please let me know what might be helpful. 

Thank you for your help.

crasu, 2023-11-27:
Does the output show up on the sticks GUI as well or just in homeassistant?

I have a TL-X inverter which gets its internal power from the DC side. It turns off at night so I do not get any power readings.

truwer, 2023-11-29:
Hi,
the values from GUI and HA looks like exactly the same. I created another screenshot. I also added a riemann helper (PV Energy Riemann) to sum the actual power ("PV Input Power (Solar)") to energy . This looks like the values I expected. Only rising when the sun comes up, no changes in the night.

I recognized that the faulty "PV Energy (today)" rises parallel to the "PV AC Charge Energy (Today)". 

**Does this mean there is energy from the grid used to charge the battery, which in the end is counted as "PV Energy"?**

[image]

crasu, 2023-12-08:
This is more an inverter questions than an issue with the firmware. Please move discussions over here:
https://github.com/OpenInverterGateway/OpenInverterGateway/discussions

closed 2023-12-08 by crasu