OpenInverterGateway/OpenInverterGateway #290: Error: Unknown Shine Stick

Folllast, 2023-12-09:
# What happened?
My Inverter MIC2000 is running. The green leaf is lighted and i see the power etc. in de Display. But there are no values on the GroStick. every value is 0 and the status is 0 (Waiting). I changed nothin on the setup.

# Stick type

What Stick type are you using?
- USB Shine Stocl

Which Inverter are you using?
- MIC-2000-TL-X

# Log output
Please copy and paste any relevant log output from /debug and the serial console (You have to compile the firmware with the relevant settings in Config.h to get the output).

```
Setup()
Opened serial web server on http://(IP unset):8080
WIFI connected...yeey :)
MqttServer: 192.168.0.3
MqttPort: 1883
MqttTopic: energy/solar
Error: Unknown Shine Stick
MqttServer: 192.168.0.3
MqttUser: homeassistant
MqttTopic: energy/solar
Attempting MQTT connection...connected
Subscribed to energy/solar/command/#
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick
Error: Unknown Shine Stick

```

crasu, 2023-12-10:
> .... I changed nothin on the setup.
> 
Are you using the precompiled firmware? Did you do an update and was it working before?

Folllast, 2023-12-10:
Ir was working for 2 months. I installed the precompiled firmware.
I am pretty sure that the stick does not have an automatic update function and that the inverter cannot update itself via the stick.
A few days ago it snowed here and the inverter was off for several days. Now it is running again, but the stick is not supplying any data.

aviadoffer, 2024-01-02:
Today I had a similar issue with the same error message. this was working for the past 4 months with no issue. nothing changed as far as I can tell, I didn't upgrade etc.
My inverter is a MIN 6000TL-X.  My logs already rolled but I saw the same Unknown shine stick. about 20 min later everything started working again. I did try few reboots and physical disconnect which didn't help but on it's own it all the sudden connected fine:
MqttUser: 
MqttTopic: energy/solar
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
MQTT Connect failed
ReadData() successful
publish MQTT message... not connected
MqttServer: 10.1.2.3

I'm not sure how to enable ssh to get more logs...

crasu, 2024-02-11:
Sry for my late response. Please make sure that serial debug is disabled. As it puts out log messages on the same serial port which is used for the modbus connections (I always wondered how this is working at all).

There is a baudrate autodetection feature in the code I do not really trust. We could try to disable this. Anyone willing to test that?

theguardian85, 2024-03-11:
I've just recompiled the code for a D1 board, uploaded it and it working again. More than happy to do it again without the baudrate auto detection.

Fyi: While recompiling, PlatformIO did had issues with line 37 in ShineMqtt: return HOSTNAME + String(id & 0xffffffff); -> the String(id & 0xffffffff); gives an error.

Folllast, 2024-03-18:
> Sry for my late response. Please make sure that serial debug is disabled. As it puts out log messages on the same serial port which is used for the modbus connections (I always wondered how this is [...]

1. Where should I disable the serial debug? in the Inverter or on the Stick? is there a register to disable it via stick?
2. If there is a fixed baudrate in the WR, maybe a fixed baudrate in the Software should be more stable.

crasu, 2024-03-20:
1. In Config.h - Debug is disabled by default if you use a precompiled image
2. I made it possible to set a fixed baud rate via the Config.h and attached precompiled firmware images to this post. Feel free to test them - feedback welcome.
[Uploading firmware-disable-auto-detection.tar.gz…]()

Neoster, 2024-05-05:
Today I am experiencing the same 'Unknown Shine Stick' message for my Growatt MIN2500 TL-X.

I am running tests on the ShineWifi-X as well as on a brand new NodeMCU V2.
Both seem to fail communicating with the inverter.

This is my log:

- Opened serial web server on http://(IP unset):8080
- Setting up WiFiManager menu
- WIFI connected...yeey :)
- MqttServer: 192.168.2.250
- MqttPort: 1883
- MqttTopic: energy/solar
- Modbus: init with slave id: 3
- Modbus: init response: 0xE2
- Modbus error: Response Timed Out
- Error: Unknown Shine Stick
- MqttServer: 192.168.2.250
- MqttUser: homeassistant
- MqttTopic: energy/solar
- Attempting MQTT connection...connected
- Subscribed to energy/solar/command/#

I found another Issue about this, suggesting I should change the slave id to 4. I tried this without success.

I am compiling using the ArduinoIDE.

By the way, how do I know which versions of modbus I should compile?

Any suggestions are welcome!

gekkehenkie11, 2024-06-27:
I have the same error too. Lucikly I have several wifi sticks and I indeed note that there are different hardware revisions. It seems it only works with this one: https://github.com/OpenInverterGateway/OpenInverterGateway/blob/master/Doc/ShineWiFi-X/ShineWiFi-X_PCB_Top.jpg

that has that "XR21V1410" chip (the chip closest to the USB connector on the photo). Going to try with the 'right' one later today

crasu, 2024-07-29:
Closing this due to missing response from original poster

closed 2024-07-29 by crasu