OpenInverterGateway/OpenInverterGateway #294: Huge first value

dirkvdb, 2023-12-17:
# What happened?
The first "OutputPower" value that gets published in the morning is sometimes the huge 214748364,7 value which is max int32 / 10. 

# Stick type
- Shine X

# Inverter
- MOD-8000-TL3-XH

DutchKillerbee, 2023-12-17:
Hello diskvdb

this was also mentioned in issue https://github.com/OpenInverterGateway/OpenInverterGateway/issues/246
Issue is that your initial (first) value is a negative value and the register is not red as a signed integer but as an unsigned integer.
So the negative value gives an overflow (= 214748364,7)

dirkvdb, 2023-12-17:
Ok, I will change the type of the `"PVOutputPower"` input register to `SIZE_32BIT_S` and see if behaves correctly

crasu, 2023-12-18:
fixed by #295

closed 2023-12-18 by crasu