OpenInverterGateway/OpenInverterGateway #299: Which EEPROM.h use for this?

lakusz1github, 2024-01-02:
# What happened?
i cant build :(

# Stick type

What Stick type are you using?
- d1

# Log output
In file included from .pio\libdeps\d1\StreamUtils\src/StreamUtils.hpp:17,
                 from .pio\libdeps\d1\StreamUtils\src/StreamUtils.h:5,
                 from C:\Users\xxx\.platformio\platforms\espressif8266\builder\SRC\ShineWiFi-ModBus\ShineMqtt.cpp:5:
.pio\libdeps\d1\StreamUtils\src/StreamUtils/Streams/EepromStream.hpp:11:10: fatal error: EEPROM.h: No such file or directory

****************************************************************
* Looking for EEPROM.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:EEPROM.h"
* Web  > https://registry.platformio.org/search?q=header:EEPROM.h
*
****************************************************************

   11 | #include <EEPROM.h>
      |          ^~~~~~~~~~
compilation terminated.

```
Insert logs here
```

lakusz1github, 2024-01-02:
i tested : [mbed-jackb/EEPROM](https://registry.platformio.org/libraries/mbed-jackb/EEPROM)
[mbed-wodji/EEPROM](https://registry.platformio.org/libraries/mbed-wodji/EEPROM)
and for fun:
[fabiobatsilva/ArduinoFake](https://registry.platformio.org/libraries/fabiobatsilva/ArduinoFake)

crasu, 2024-02-07:
You should not need any of those libs. Platform io should automatically download all required libs for you. Are you using a current version of pio? platformio-6.1.13 works on the ci.