OpenInverterGateway/OpenInverterGateway #303: Strange values for TodayGenerateEnergy

truwer, 2024-01-08:
# What happened?
There are days when the values for TodayGenerateEnergy does not fit. It looks like the PV generates energy at night. No energy is generated when its dark, as I expect. If you compare the value with the two strings PV1 and PV2 it does fit on those days. Why does it look like this?

Side question: Why does the Inverter unload the batterie below 10% on some days. (I guess not a data logger issue, but maybe the experts have an idea)

What Stick type are you using?
- Shine X

Which Inverter are you using?
- SPH-8000 TL3 BL-UP

I grab the mqtt message in Home Assistant and you can see the visualized result here. The values are the same as in the web server itself.

[image]

daniel-iiv, 2024-01-20:
I have the same problem. According to the value in TodayGenerateEnergy, electricity is generated at night, but logically this cannot be correct.
However, the value in the PV1 and PV2 strings seems to be correct. I have not yet found an answer to this either.

I am using Shine X together with SPH 10000TL3 BH-UP.

miguent, 2024-02-02:
Hi.

Exactly the same issue hiere, using the SPH 10000TL3 BH-UP.
I am using an NodeMCU which is working perfectly fine, except this strange value.

Thanks for the hint with using the values for PV1 and PV2. This I will use as fallback at first place.

Thanks for this awesome project!

Best regards,
Michael

crasu, 2024-02-08:
Could someone with a BH-UP query the values directly via the modbus page? I want to make sure that it is not a strange memory corruption thing. @truwer @daniel-iiv @miguent I guess the PV power values stay at 0 during the night.

truwer, 2024-02-08:
Hi @crasu I will test anything I can. Could you please explain what is meant with the modbus page directly? I just have the flashed Growatt stick and a stick with the original firmware. Is any other hardware required? 

These are the values from the flashed stick from 7:21 am and it is still fully dark.

`{"InverterStatus":6,"InputPower":0,"PV1Voltage":0,"PV1InputCurrent":0,"PV1InputPower":0,"PV2Voltage":0,"PV2InputCurrent":0,"PV2InputPower":0,"OutputPower":250.8,"GridFrequency":49.97,"L1ThreePhaseGridVoltage":396.7,"L1ThreePhaseGridOutputCurrent":0.6,"L1ThreePhaseGridOutputPower":496,"L2ThreePhaseGridVoltage":393,"L2ThreePhaseGridOutputCurrent":0.7,"L2ThreePhaseGridOutputPower":0,"L3ThreePhaseGridVoltage":397.1,"L3ThreePhaseGridOutputCurrent":0.6,"L3ThreePhaseGridOutputPower":0,"TodayGenerateEnergy":1.4,"TotalGenerateEnergy":7742,"TWorkTimeTotal":2.7725624e7,"PV1EnergyToday":0,"PV1EnergyTotal":2909.9,"PV2EnergyToday":0,"PV2EnergyTotal":5398.7,"PVEnergyTotal":8308.6,"InverterTemperature":31.3,"TemperatureInsideIPM":29.6,"BoostTemperature":30.3,"DischargePower":410,"ChargePower":0,"BatteryVoltage":154.7,"SOC":12,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":410,"INVPowerToLocalLoadTotal":410,"BatteryTemperature":236,"BatteryState":6,"EnergyToUserToday":0,"EnergyToUserTotal":1061.8,"EnergyToGridToday":0,"EnergyToGridTotal":5094.9,"DischargeEnergyToday":2.6,"DischargeEnergyTotal":1177.3,"ChargeEnergyToday":0,"ChargeEnergyTotal":1431.9,"LocalLoadEnergyToday":2.6,"LocalLoadEnergyTotal":4016,"ACChargeEnergyToday":0.8,"ACChargeEnergyTotal":462.9,"Mac":"4C:75:25:2C:45:E5","Cnt":211162}`

miguent, 2024-02-08:
Hi @crasu ,

these are my current values:

```
{
    "InverterStatus": 6,
    "InputPower": 0,
    "PV1Voltage": 0,
    "PV1InputCurrent": 0,
    "PV1InputPower": 0,
    "PV2Voltage": 0,
    "PV2InputCurrent": 0,
    "PV2InputPower": 0,
    "OutputPower": 0,
    "GridFrequency": 49.97,
    "L1ThreePhaseGridVoltage": 409.1,
    "L1ThreePhaseGridOutputCurrent": 0.6,
    "L1ThreePhaseGridOutputPower": 446.7,
    "L2ThreePhaseGridVoltage": 407.9,
    "L2ThreePhaseGridOutputCurrent": 0.6,
    "L2ThreePhaseGridOutputPower": 0,
    "L3ThreePhaseGridVoltage": 410.5,
    "L3ThreePhaseGridOutputCurrent": 0.6,
    "L3ThreePhaseGridOutputPower": 0,
    "TodayGenerateEnergy": 2.6,
    "TotalGenerateEnergy": 598.4,
    "TWorkTimeTotal": 6276094,
    "PV1EnergyToday": 0,
    "PV1EnergyTotal": 114.3,
    "PV2EnergyToday": 0,
    "PV2EnergyTotal": 152.1,
    "PVEnergyTotal": 266.4,
    "InverterTemperature": 28.2,
    "TemperatureInsideIPM": 28.1,
    "BoostTemperature": 26.5,
    "DischargePower": 0,
    "ChargePower": 0,
    "BatteryVoltage": 153.5,
    "SOC": 10,
    "ACPowerToUser": 360,
    "ACPowerToUserTotal": 360,
    "ACPowerToGrid": 0,
    "ACPowerToGridTotal": 0,
    "INVPowerToLocalLoad": 360,
    "INVPowerToLocalLoadTotal": 360,
    "BatteryTemperature": 15.7,
    "BatteryState": 6,
    "EnergyToUserToday": 7.7,
    "EnergyToUserTotal": 1400.1,
    "EnergyToGridToday": 0,
    "EnergyToGridTotal": 3.8,
    "DischargeEnergyToday": 0,
    "DischargeEnergyTotal": 130.2,
    "ChargeEnergyToday": 0,
    "ChargeEnergyTotal": 149.4,
    "LocalLoadEnergyToday": 7.6,
    "LocalLoadEnergyTotal": 1654.9,
    "ACChargeEnergyToday": 1.3,
    "ACChargeEnergyTotal": 169.8,
    "Mac": "48:3F:DA:65:5E:C0",
    "Cnt": 17480
}
```

Same question as @truwer - can I query these values somehow different to support more here?

Cheers,
Michael

crasu, 2024-02-11:
@truwer @miguent Go to http://<stick-ip>/postCommunicationModbus and query registers 53 (32 bit).

miguent, 2024-02-13:
Hi @crasu ,

I queried register 53. But as I am not sure if it is a holding or input register, these are both values:

Input Register (53, 32 Bit): Read 32b Input register 53 with value 14
Holding Register (53, 32 Bit): Read 32b Holding register 53 with value 300225216

Value for TodayGenerateEnergy at that point of time was: 1.4

Thanks and best regards,
Michael

plieven, 2024-06-27:
@miguent @truwer @crasu the behavior is correct if there is a DC battery is attached to the inverter and the battery discharges at night. Total/Todaygenerateenergy accounts the energy that was generated by the inverter regardless if it was sourced by PV or battery. @crasu maybe it would be helpful to ask more details about the setup when opening an issues. Connected battery system e.g.

miguent, 2024-07-03:
@plieven ah ok, that totally makes sense. But shouldn't in that case the DischargeEnergyToday reflect that value somehow? Checking my example values from above, TotalGenerateEnegery has value 2.6, but PV1EnergyToday, PV2EnergyToday and DischargeEnergyToday have a value of 0.

plieven, 2024-07-03:
@miguent If you refer to the Json Output from above there is a DischargeEnergyToday of 2.6 which also makes no sense. Maybe my assumption about the TodayGenerateEnergy is wrong. What also makes no sense is that there is only a non zero value on L1ThreePhaseOutputPower and zero on all other phases. Is this an off-grid setup? In this case the inverter would have to compensate all reactive power. And in this case DischargeEnergyToday could of course be much higher than TodayGenerateEnergy.