OpenInverterGateway/OpenInverterGateway #310: Battery Status not available with Growatt mod 9000tl3-xh

newokatrix, 2024-01-23:
# What happened?
All battery related values are empty. I have used the compiled version you have hosted here.

# Stick type
What Stick type are you using?
- Shine X 

Which Inverter are you using?
- MOD 9000 TL3-XH

# JSON output
below the copy of the JSON string. 
```
{"InverterStatus":1,"InputPower":973.7,"PV1Voltage":251.6,"PV1InputCurrent":1.5,"PV1InputPower":377.4,"PV2Voltage":372.7,"PV2InputCurrent":1.6,"PV2InputPower":596.3,"OutputPower":371.8,"GridFrequency":50.03,"L1ThreePhaseGridVoltage":238.3,"L1ThreePhaseGridOutputCurrent":0.7,"L1ThreePhaseGridOutputPower":166.8,"L2ThreePhaseGridVoltage":240.9,"L2ThreePhaseGridOutputCurrent":0.7,"L2ThreePhaseGridOutputPower":168.6,"L3ThreePhaseGridVoltage":238.3,"L3ThreePhaseGridOutputCurrent":0.8,"L3ThreePhaseGridOutputPower":190.6,"TodayGenerateEnergy":2.5,"TotalGenerateEnergy":5640.7,"TWorkTimeTotal":2.3860614e7,"PV1EnergyToday":2.5,"PV1EnergyTotal":2442.2,"PV2EnergyToday":2.9,"PV2EnergyTotal":3374.9,"PVEnergyTotal":5817.1,"InverterTemperature":43.6,"TemperatureInsideIPM":29.2,"BoostTemperature":28.3,"DischargePower":0,"ChargePower":0,"BatteryVoltage":0,"SOC":0,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":0,"BatteryState":0,"EnergyToUserToday":0,"EnergyToUserTotal":0,"EnergyToGridToday":0,"EnergyToGridTotal":0,"DischargeEnergyToday":0,"DischargeEnergyTotal":0,"ChargeEnergyToday":0,"ChargeEnergyTotal":0,"LocalLoadEnergyToday":0,"LocalLoadEnergyTotal":0,"ACChargeEnergyToday":0,"ACChargeEnergyTotal":0,"Mac":"xxxxx","Cnt":1490}
```

I would be happy if you have any hints or tips to fix this!

crasu, 2024-02-08:
@newokatrix This is the same issue as #247 and #197. As long as no one with a TL-XH inverter provides a PR the values will sadly stay unsupported.

closed 2024-02-08 by crasu

newokatrix, 2024-02-08:
@crasu Thank you for your feedback. That was very helpful. I used the work fom [WaizenErnter](https://github.com/WaizenErnter). Do compile it. Will upload the Firmware here ... THX again!

plieven, 2024-06-27:
@crasu @newokatrix I have support for TL-XH inverters complete. Just have to find time to prepare the pull requests.

newokatrix, 2024-06-27:
> @crasu @newokatrix I have support for TL-XH inverters complete. Just have to find time to prepare the pull requests.

Thank you, i have already uploaded a working version ( see comment above ) which is running stable for month ... but happy to test your new update :)

plieven, 2024-06-27:
@newokatrix is there source available for the working version?
all registers for the TL-XH inverters are different including the holding registers for controlling the battery (charge/discharge power, timeslots etc.)

newokatrix, 2024-06-27:
> @newokatrix is there source available for the working version? all registers for the TL-XH inverters are different including the holding registers for controlling the battery (charge/discharge power, timeslots etc.)

Basicly i just took your code and updated the config.h to GROWATT_MODBUS_VERSION to 6000. Nothing more was needed.

plieven, 2024-06-27:
@newokatrix I wonder which modbus documentation was used for implemeting the BP series support. According to modbus spec 1.24 the battery information is at different registers. the registers above 4000 which are used in the BP implementation are not documentated there.

Could you give me a favor and check if modbus input register 3171 (16-bit) reflects your SOC?

If you want to experiment you could also clone https://github.com/plieven/OpenInverterGateway/ and check if it works for you. Please be aware that this is my testing code. It does only complile for protocol version 3000 at the moment as I was going to implement unit testing which meant I had to move a lot of source files. Please be also aware that it has NTP support build in. Please adjust the timezone in Config.h if necessary.

newokatrix, 2024-06-27:
@plieven 
Sure here is the result: 
Read 16b Input register 3171 with value 99 which equal to my SOC value in the jason output

plieven, 2024-06-27:
@newokatrix this is what i expected. thats the official register for the SOC for TL-XH inverters.

@rostwolke can you advice which modbus reference you used when implementing the BP series support?