OpenInverterGateway/OpenInverterGateway #314: Protocoll Error in Mosquitto

muhkuh666, 2024-02-10:
# What happened?
Also tell us, what did you expect to happen? Tell us what you see!

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MIC-

# Log output
Mosquitto:

2024-02-10 13:21:29: New connection from 192.168.2.153:58641 on port 1883.
2024-02-10 13:21:29: Client <unknown> disconnected due to protocol error.

```
MqttServer: 192.168.2.93
MqttUser: growatt
MqttTopic: 
Attempting MQTT connection...failed, rc=-4 try again in 5 seconds
MQTT Connect failed
ReadData() successful
publish MQTT message... not connected

```
How can I fix the protocoll error?

crasu, 2024-02-11:
Do you have more than one inverter connected to the broker? Which mosquitto version are you using?

muhkuh666, 2024-02-12:
No

muhkuh666, 2024-02-12:
Found it made a mistake in config of Stick

closed 2024-02-12 by muhkuh666