OpenInverterGateway/OpenInverterGateway #330: Can't get any data from 10KTL3-X inverter.

irishgreencitrus, 2024-04-03:
# What happened?
Tried using the stock firmware, however MQTT failed to initialise, so I had to compile the source myself with MQTT disabled.
However, even with this (and trying all three MODBUS versions) I could only get the output `ReadData() NOT successful`.

Flashed with same USB Cable I used to plug into the inverter, I know that it isn't faulty.

Serial and USB debugging are also disabled, as I thought that could be the issue.

# Stick type

What Stick type are you using?
- Nodemcu32

Which Inverter are you using?
- MOD 10KTL3-X

# Log output

```
Opened serial web server on http://0.0.0.0:8080
Configuring WDT...
Setting up WiFiManager menu
WIFI connected...yeey :)
ShineWiFi-X (USB) found
Modbus: read Segment from 0x00 with len: 50 ...failed
ReadData() NOT successful
... (failure continues forever)
```

crasu, 2024-04-04:
Please double check that your Nodemcu32 does use a supported usb2serial chip. See the readme for more details which chips are supported. The nodemcu32(s) I have mostly use CP21XX chips which are not supported.

closed 2024-04-04 by crasu

irishgreencitrus, 2024-04-04:
Yes you are correct I have the CP2102 chip. A shame that it's not supported. Is it possible to look into adding support (I'm very happy to contribute this back to the original project if it works).