OpenInverterGateway/OpenInverterGateway #338: Growatt 1500-S inverter support

danielschwab, 2024-04-29:
I flash a ShineWifi-S stick with the precompiled version of OpenInverterGateway which worked just fine. After connecting the stick via the serial connector to a Growatt 1500-S inverter the webinterface only shows zero values (See log output).

Is the Growatt 1500-S inverter supported?  
I found another repository (https://github.com/rixvet/GrowattReader) which supports this inverter and it seems the inverter doesn't use the modbus interface.  

# Stick type
ShineWifi-S

# Which Inverter are you using?
 Growatt 1500-S

# Log output
```
.. skipping 693508 bytes of log - no longer in buffer ..
adData() NOT successful
ReadData() NOT successful
```

danielschwab, 2024-04-29:
After I did compile the firmware with **protocol v3.05** by my self everything seems to work. 

I just had an error (call of overloaded 'String(uint64_t)' is ambiguous) while compiling (PlatformIO in visual studio code). 
I changed `ShineMqtt,cpp:37` from:
`  return HOSTNAME + String(id & 0xffffffff);`
to:
`  return HOSTNAME + String((unsigned int)(id & 0xffffffff));`

This seems to be an issue with `ShineWifiS` PlatformIO environment which uses the `espressif8266` platform. The code compiles just fine with the `nodemcu-32s` environment which uses the `espressif32` platform.

crasu, 2024-06-07:
@danielschwab Just tried compiles for me just fine with current master

closed 2024-06-07 by crasu