OpenInverterGateway/OpenInverterGateway #342: Strange Values of OutputPower at the evening

krolli53, 2024-05-22:
What happened?
Mostly every evening (at around 8pm-9pm the values of "OutputPower" go high like "OutputPower: 214748364.7 W"... With my 4.5kW peak installation this might not be possible. Even though, TodayGenerateEnergy has a reasonable value of 4.1kWh.

Values have spiked since May 12th

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MOD 5000TL3-XH 5kW Solar Hybrid Inverter

[image]

[image]

munderl, 2024-06-14:
Have the save "problem" with my MOD10KTL3-XH using my own Code. It seems to be a ModBus read error. At the moment I simply discard implausible values directly after reading.

plieven, 2024-06-27:
Which firmware are you using? The output power has to be a signed 32 bit value. I think this was fixed after the last official binary firmware release.

crasu, 2024-08-08:
Looks like you are using an unsigned value instead of a signed one. Maybe you want to change OUTPUT_POWER to SIZE_32BIT_S in your custom code.

closed 2024-08-08 by crasu