OpenInverterGateway/OpenInverterGateway #344: Integration of smartmeter?

majuss, 2024-05-27:
# What happened?
Also tell us, what did you expect to happen? Tell us what you see!

I have a CHINT DTSU666 and would like to read out the "raw meter value" so just the real time value the meter shows.

What Stick type are you using?
- Shine X

Which Inverter are you using?
- SPH-10000 BL-UP
```
{
    "InverterStatus": 5,
    "InputPower": 338.9,
    "PV1Voltage": 191.1,
    "PV1InputCurrent": 1.2,
    "PV1InputPower": 218.3,
    "PV2Voltage": 174.5,
    "PV2InputCurrent": 0.8,
    "PV2InputPower": 123.6,
    "OutputPower": 305.1,
    "GridFrequency": 50.01,
    "L1ThreePhaseGridVoltage": 401.5,
    "L1ThreePhaseGridOutputCurrent": 0.7,
    "L1ThreePhaseGridOutputPower": 532.6,
    "L2ThreePhaseGridVoltage": 406.4,
    "L2ThreePhaseGridOutputCurrent": 0.7,
    "L2ThreePhaseGridOutputPower": 0,
    "L3ThreePhaseGridVoltage": 404.1,
    "L3ThreePhaseGridOutputCurrent": 0.7,
    "L3ThreePhaseGridOutputPower": 0,
    "TodayGenerateEnergy": 11.6,
    "TotalGenerateEnergy": 2586.2,
    "TWorkTimeTotal": 3.3794132e7,
    "PV1EnergyToday": 8.5,
    "PV1EnergyTotal": 1933,
    "PV2EnergyToday": 4.6,
    "PV2EnergyTotal": 970,
    "PVEnergyTotal": 2903,
    "InverterTemperature": 36.6,
    "TemperatureInsideIPM": 34.6,
    "BoostTemperature": 32.6,
    "DischargePower": 0,
    "ChargePower": 0,
    "BatteryVoltage": 0,
    "SOC": 0,
    "ACPowerToUser": 0,
    "ACPowerToUserTotal": 0,
    "ACPowerToGrid": 230,
    "ACPowerToGridTotal": 230,
    "INVPowerToLocalLoad": 100,
    "INVPowerToLocalLoadTotal": 100,
    "BatteryTemperature": 0,
    "BatteryState": 5,
    "EnergyToUserToday": 2.2,
    "EnergyToUserTotal": 4.1,
    "EnergyToGridToday": 10.1,
    "EnergyToGridTotal": 17.9,
    "DischargeEnergyToday": 0,
    "DischargeEnergyTotal": 0.1,
    "ChargeEnergyToday": 0,
    "ChargeEnergyTotal": 12.3,
    "LocalLoadEnergyToday": 4.9,
    "LocalLoadEnergyTotal": 2788.1,
    "ACChargeEnergyToday": 0.1,
    "ACChargeEnergyTotal": 115.3,
    "Mac": "48:55:19:67:90:13",
    "Cnt": 55
}
```

plieven, 2024-05-31:
If I am correct the meter values are there, but the registers are a bit misleading.
If the total system power is negative (you are exporting) you will find the meter value in ACPowerToGrid.
If you are importing (the total system power is positive) you will see this in ACPowerToUser.
The both are mutually exclusive. Only one of the registers can be non-zero.
Can you verify this?

crasu, 2024-07-29:
Closing due to missing response from original poster

closed 2024-07-29 by crasu