OpenInverterGateway/OpenInverterGateway #392: Problems with receiveing battery SOC or battery values

zeddjay9878, 2024-07-04:
# What happened?
HI there, I use the OenInvertergateway successfully since a few days, but I can´t read the SOC or any other battery status, like charging or discharging. Also on the internal webinterface I see only inverter states

# Stick type
Shine X

plieven, 2024-07-04:
@zeddjay9878  Which Inverter are you using?

zeddjay9878, 2024-07-04:
MOD-10000TL3-XH

plieven, 2024-07-04:
TL-XH inverters use different registers especially for the BDC that are currently not supported. I am currently working on getting all my changes upstream, but this will take some time for review. If you want to test and can compile an image yourself you can check my repo https://github.com/plieven/OpenInverterGateway/. Make sure to choose GROWATT_MODBUS_VERSION 3000 in Config.h.

zeddjay9878, 2024-07-04:
Ok I will give them a try

plieven, 2024-07-04:
@zeddjay9878 please let me know if it works for you.

zeddjay9878, 2024-07-05:
Good monring, sorry i don´t work, I got an OLD yaml, shinex .... this one work, I must only modify some registers to find the correct values

plieven, 2024-07-05:
What exactly does not work? I also have a ShineWifiX. Can you paste the output of /status or /metrics please.Von meinem iPhone gesendet

p4co86, 2024-07-06:
@plieven  I have tested your repo, thanks for that. Everything is working fine, as well as the soc value. Only issue i could recognize is the batteryvalue, its 2122,4V instead of 212,24V.

plieven, 2024-07-06:
@p4co86 thanks for your feedback. which battery system are you using? for growatt apx the battery voltage is correct with a multiplier of 0.1. however the modbus reference shows says its 0.01. I think we have to distinguish in the code and somehow detect the battery system.

12retord, 2024-07-21:
Hello, same problem here. Ihave also the MOD-10KTL3-XH with battery. Is there a chance that there will be a "ready-for-use"-release in the next time. I'm sadly not really able to compile the firmeware on my one. Maybe someone can give me some hints (Visual Studio Code with PIO Home already installed, but not really knowing how to work with it) Thanks for help

xieve, 2024-07-21:
I have a MOD-10KTL3-XH with battery as well, and it seems it is running v3.05 despite the QC test report being dated in 2023:
```
Read 16b Holding register 88 with value 305
```
[image]

crasu, 2024-07-29:
@zeddjay9878 The support for the MOD-10KTL3-XH is now merged. Regarding your suggestion for a precompiled firmware, the firmware version is currently not changeable at runtime. This limitation means we would need to build a separate firmware version for each inverter and stick type combination, which is not scalable.

Making the firmware version changeable at runtime would require significant modifications to the firmware. Therefore, for the foreseeable future, you will need to compile the firmware yourself.

closed 2024-07-29 by crasu

plieven, 2024-07-29:
@crasu the combination TL-X(H) with shinewifix stick is quite common. It might be a good for the spread of the project to support this. If you want I can help you with the generation of images. I only have two PR in queue (not yet submitted) that would be good for a generic image (fragment retry logic and syslog support). Maybe this would be a good point to cut off a new release.