OpenInverterGateway/OpenInverterGateway #402: Can I use original and custom stick at the same time using stick pins?

scorninpc, 2024-07-31:
# What happened?
I know this is not fw related question, but can I use this ports connect to my ESP, and use use original and my custom at the same time? 

I know that I can read pin 2 and 3 from DB9, but it will more easy solde this pins, and i cannot find datasheet of this one (i'm new one in solar things, i think i dont get the ways to find the things yet)

[image]

What Stick type are you using?
- Shine S

Which Inverter are you using?
- CSI-5K-MTL

crasu, 2024-08-08:
On a growatt inverter (e.g. 600tl-x) with USB this is possible. However it you have to connect one stick via usb and the other one via an rs485 converter. This kind of works most of the time.

However I do not know if this works for a CSI-5K-MTL inverter as this one does not seem to be procduced by growatt(?).

closed 2024-08-08 by crasu

scorninpc, 2024-08-08:
Hi, thanks for your attention. I thought the stick would be mandatory for this type of solution, not the inverter. Anyway, I will take out the stick, open it and do some tests. Thanks