OpenInverterGateway/OpenInverterGateway #411: No battery data

realfldh, 2024-08-25:
# What happened?
Basis data are shown - Jipiiie!
But not any battery information - a comms issue to the battery or not supported anyway?

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MOD 5000TL3-XH 
- 
# Log output
[image]
```
Insert logs here
```

Stephan3, 2024-09-06:
Did you use the release version for your flash?

JonasGroeger, 2024-09-10:
The precompiled versions won't work. You need to compile youtself and set `GROWATT_MODBUS_VERSION` to 6000 in `config.h`.

Untested, but should work, since I have the same setup and the registers work using http:

```
$ http --form 'growatt-10k/postCommunicationModbus_p' val= operation=R registerType=I type=16b reg=4014
Read 16b Input register 4014 with value 52

$ http --form 'growatt-10k/postCommunicationModbus_p' val= operation=R registerType=I type=32b reg=4023
Read 32b Input register 4023 with value 0

$ http --form 'growatt-10k/postCommunicationModbus_p' val= operation=R registerType=I type=32b reg=4021
Read 32b Input register 4021 with value 5340

```

Introduced in afb71f4ca3fa533ca887083bd29d7665f5dcf778.

JonasGroeger, 2024-09-20:
Using the latest version (3.0.1) and using 6000 as mentioned above, I get

[image]

SOC works, Charge and Discharge works.

crasu, 2024-09-21:
@realfldh You can try one of these TLXH images: https://github.com/crasu/Growatt_ShineWiFi-S/releases/tag/release%2Fv3.1.0

closed 2024-09-21 by crasu