OpenInverterGateway/OpenInverterGateway #416: Data to EVCC

f1nch87, 2024-09-02:
I flashed the precompiled wifix.bin. All is working but i only have one topic in mqtt with all its data in it.

I want to send the data to evcc.

Anyone has an example how he sends his data to evcc?

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MOD-4000TLXH

Thx

Stephan3, 2024-09-06:
EVCC need to parse the JSON. I needed to do it in iobroker by script for example.

nacho26, 2024-09-09:
Yes I get to work with this configuration in my evcc.yaml:

- name: solar
  type: custom
  power:
    source: http
    uri: http://1.1.1.1/status
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    jq: .OutputPower
    timeout: 5s

plieven, 2024-09-09:
@nacho26 If you are using the latest master you can query individual values directly: http://1.1.1.1/value/OutputPower

Unfortunately no release binary yet.

thobastian, 2025-01-17:
If I use the template von Nacho26, I get an error:cannot create meter 'solar': cannot create meter type 'custom': decoding failed due to the following error(s): '' has invalid keys: headers, jq, method, source, timeout, uri

Whats wrong with it?

my version:
name: solar
  type: custom
  power:
  source: http
  uri: http://192.168.178.119/status
  method: GET # default HTTP method
  headers:
  - content-type: application/json
  jq: .OutputPower
  timeout: 5s

thobastian, 2025-01-17:
and what are my work in evcc.yaml ?

crasu, 2025-05-07:
not a bug

closed 2025-05-07 by crasu