OpenInverterGateway/OpenInverterGateway #425: broken lib_deps: https://github.com/rob040/ModbusMaster#16d5f1f1cb4e1c9a7943425557a09ac9e89d624c no longer available?

physiko, 2024-09-26:
# What happened?
I wanted to compile myself to adjust the Modbus Protocol support. Compilation is not possible because the ModbusMaster lib from rob040 dissapeared:
https://github.com/rob040/ModbusMaster#16d5f1f1cb4e1c9a7943425557a09ac9e89d624c

# Stick type
general problem in compiling

# Log output
Page not found 404 message see here: https://github.com/rob040/ModbusMaster#16d5f1f1cb4e1c9a7943425557a09ac9e89d624c

plieven, 2024-09-26:
@physiko thanks for reporting.
@crasu  can you add a copy of the Modbus Library to the OpenInverterGateway account and link it in the requirements?
If you don't have a copy of the ModbusMaster with timeout support I uploaded a copy here:

https://github.com/plieven/ModbusMaster#16d5f1f1cb4e1c9a7943425557a09ac9e89d624c

crasu, 2024-09-26:
I will do so tonight

https://github.com/plieven/ModbusMaster#16d5f1f1cb4e1c9a7943425557a09ac9e89d624c

crasu, 2024-09-26:
@physiko thank you for the report. Should be fixed with the latest master

closed 2024-09-26 by crasu