OpenInverterGateway/OpenInverterGateway #433: Cannot Change Max Power Limit

InvertedMinor, 2024-11-07:
# What happened?

I cannot change  the Growatt Max Power Limit (needed to limit the production in the summer to avoid fees from the electricity supplier). This is what is do in home assistant:

Created a MQTT switch:

`- name: "Max Power Limit"
    state_topic: "energy/solar/OutputMaxpowerLimited"
    command_topic: "energy/solar/OutputMaxpowerLimited/set"
    qos: 1
    payload_on: "4000"      
    payload_off: "0"         
    state_on: "4000"
    state_off: "0"
    optimistic: false
    retain: true `

When 'pressed' the following is visible in MQTT explorer:

[image]

But the mqtt sensor value is still at 6000 unchanged?

`{"InverterStatus":1,"InverterRunState":0,"PVTotalPower":0,"PV1Voltage":140.7,"PV1InputCurrent":0,"PV1Power":0,"PV2Voltage":141.8,"PV2InputCurrent":0,"PV2Power":0,"PV3Voltage":0,"PV3InputCurrent":0,"PV3Power":0,"PV4Voltage":0,"PV4InputCurrent":0,"PV4Power":0,"SystemOutputPower":0,"ReactivePower":141.3,"OutputPower":0.6,"GridFrequency":49.98,"L1ThreePhaseGridVoltage":227.9,"L1ThreePhaseGridOutputCurrent":0.4,"L1ThreePhaseGridOutputPower":91.1,"L2ThreePhaseGridVoltage":227.9,"L2ThreePhaseGridOutputCurrent":0.4,"L2ThreePhaseGridOutputPower":91.1,"L3ThreePhaseGridVoltage":225.1,"L3ThreePhaseGridOutputCurrent":0.4,"L3ThreePhaseGridOutputPower":90,"RSThreePhaseGridVoltage":395.4,"STThreePhaseGridVoltage":392.3,"TRThreePhaseGridVoltage":391.3,"TotalForwardPower":1482.1,"TotalReversePower":0,"TotalLoadPower":1482.1,"WorkTimeTotal":1.0932251e7,"TodayGenerateEnergy":1.7,"TotalGenerateEnergy":4950.2,"PVEnergyTotal":5076.9,"PV1EnergyToday":0.9,"PV1EnergyTotal":3352.7,"PV2EnergyToday":0.3,"PV2EnergyTotal":1724.2,"PV3EnergyToday":0,"PV3EnergyTotal":0,"TodayEnergyToUser":44.9,"TotalEnergyToUser":1452.1,"TodayEnergyToGrid":0,"TotalEnergyToGrid":3315.3,"TodayEnergyOfUserLoad":46.1,"TotalEnergyOfUserLoad":3213.7,"PV4EnergyToday":0,"PV4EnergyTotal":0,"PVEnergyToday":1.2,"DeratingMode":0,"PVISOValue":5451,"RDCICurr":0,"SDCICurr":0,"TDCICurr":0,"GFCICurr":0,"TotalBusVoltage":609.1,"InverterTemperature":31.2,"TemperatureInsideIPM":46.5,"BoostTemperature":29.2,"CommunicationBoardTemperature":35.9,"PBusInsideVoltage":305.3,"NBusInsideVoltage":303.7,"InverterOutputPFNow":20000,"RealOutputPercent":0,"OutputMaxpowerLimited":6000,"InverterFaultMaincode":0,"InverterWarnMaincode":0,"AFCIStatus":0,"InvStartDelayTime":60,"BDCConnectState":0,"DryContactState":0,"SelfUsePower":0,"SystemEnergyToday":1.8,"DischargeEnergyToday":0,"DischargeEnergyTotal":0,"ChargeEnergyToday":0,"ChargeEnergyTotal":0,"ACChargeEnergyToday":0,"ACChargeEnergyTotal":0,"SystemEnergyTotal":5144.9,"SelfOutputEnergyToday":1.8,"SelfOutputEnergyTotal":1829.5,"Priority":0,"BDCDeratingMode":0,"BDCSysState":0,"BDCSysMode":0,"BDCFaultCode":0,"BDCWarnCode":0,"BDCBatteryVoltage":0,"BDCBatteryCurrent":0,"BDCStateOfCharge":0,"BDCTotalBusVoltage":0,"BDCOnTheBusVoltage":0,"BDCBuckBoostCurrent":0,"BDCLlcCurrent":0,"BDCTemperatureA":0,"BDCTemperatureB":0,"BDCDischargePower":0,"BDCChargePower":0,"BDCDischargeEnergyTotal":0,"BDCChargeEnergyTotal":0,"ActivePowerRate":100,"BDCDischargePowerRate":100,"BDCDischargeStopSOC":10,"BDCChargePowerRate":100,"BDCChargeStopSOC":100,"BDCChargeACEnabled":0,"Mac":"E8:9F:6D:83:98:D9","Cnt":51780,"Uptime":199823,"WifiRSSI":-64,"HeapFree":24664,"HeapMaxAlloc":23832,"HeapMinFree":22256,"HeapFragmentation":4}`

# Stick type

What Stick type are you using?
- Shine X

Flashed with firmware -ShineWifiX-TLXH.bin

Which Inverter are you using?

Growatt MOD 6.000 TL3-XH(BP)

InvertedMinor, 2024-11-10:
Resolved: I have used the wrong topic, it must be activerate.

This is now working:

`- alias: Growatt set active rate
    description: "set active rate %"
    triggers:
      - trigger: event
        event_type: ""
    conditions: []
    actions:
      - action: mqtt.publish
        data:
          qos: "1"
          topic: energy/solar/command/power/set/activerate
          payload: |-
            {
              "correlationId": "set_activerate",
              "value": "{{ states('input_number.growatt_power')| round(0) }}"
            }
    mode: single`

closed 2024-11-10 by InvertedMinor

HENN0, 2025-04-21:
@InvertedMinor 
may you can share your complete home assistant config for this?
I cant get it to work

InvertedMinor, 2025-05-03:
@HENN0  of course here is my full solution :

`
    
    
    - name: "Power Rate"
      unique_id: growatt_activerate
      state_topic: "energy/solar/command/power/get/activerate"
      value_template: "{{ value_json.value }}"
      command_topic: "energy/solar/command/power/set/activerate"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: "%"
      min: 0
      max: 100
      step: 1
      optimistic: true
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{ '1' if value_json.InverterStatus|int(0) >= 0 else '0' }}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"
          
    - name: "Charge Power Rate"
      unique_id: bdcchargepowerrate
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCChargePowerRate'] }}"
      command_topic: "energy/solar/command/bdc/set/chargepowerrate"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 0
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"      
          
    - name: "Discharge Power Rate"
      unique_id: bdc_discharge_power_rate
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCDischargePowerRate'] }}"
      command_topic: "energy/solar/command/bdc/set/dischargepowerrate"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 0
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"      
          
    - name: "Charge Stop SoC"
      unique_id: bdc_charge_stop_soc
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCChargeStopSOC'] }}"
      icon: mdi:battery-plus-variant
      command_topic: "energy/solar/command/bdc/set/chargestopsoc"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 0
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"

    

    - name: "Discharge Stop SoC"
      unique_id: bdc_discharge_stop_soc
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCDischargeStopSOC'] }}"
      icon: mdi:battery-minus-variant
      command_topic: "energy/solar/command/bdc/set/dischargestopsoc"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 10
      max: 100
      step: 1
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"

`