OpenInverterGateway/OpenInverterGateway #435: Problem with Shine-S stick. ReadData() NOT successful

montero83, 2024-11-15:
# What happened?
Hi , after I flashed a stick , i can;t connect to the modbus. I have 2 inverters but second one has ShineX stick and it works fine, but this not. Can you help me what can I do or someone had a simillar situation.
# Stick type

What Stick type are you using?

- Shine S

Which Inverter are you using?
- Growatt 5000TL3-S

# Log output
[image]

mathias172, 2025-02-02:
Try setting another Modbus Version in Config.h and try again. Solved my problem with Shine-S stick and 3000S Inverter.

elFarto, 2025-02-08:
> Try setting another Modbus Version in Config.h and try again. Solved my problem with Shine-S stick and 3000S Inverter.

I have the same issue here. What version did you end up using?

mathias172, 2025-02-08:
3000S is Modbus 3.05, so:
#define GROWATT_MODBUS_VERSION 305

elFarto, 2025-02-08:
Thanks, that setting worked!