OpenInverterGateway/OpenInverterGateway #444: Befehle an den Growatt senden

tedesco1968, 2024-12-25:
wie funktioniert die Leistungsbegrenzung (ermöglicht Null-Export).
Was muss man wie machen? Wie kann ich Befehle an den Growatt Wechselrichter senden? 
Gehe in den Standby  oder nutze nur die Battery etc.

tedesco1968, 2025-01-03:
Hallo,
habe jetzt versucht über den knopf RW Modbus befehle an den Growatt Sph 3Tl zu senden, z.B. an die Adresse 00 für Einschalten (1) bzw. Standby (0) zu senden, was auch funktioniert.
Wie ist es möglich Befehle über den Explorer z.B Http://IP-Adresse/postCommunicationModbus wie setzt sich die Zeile zusammen wenn ich in die Adresse 00 eine 1 reinschreiben will.

tedesco1968, 2025-01-04:
hat sich erledigt

distancerunner, 2025-01-28:
Kannst du bitte noch mitteilen wie du es gelöst hast.

Can you please give us any idea, how you solved this issue?

distancerunner, 2025-01-28:
#433 here you are. 
using topic: energy/solar/command/power/set/activerate

closed 2025-01-31 by crasu