OpenInverterGateway/OpenInverterGateway #463: Odd temperature data

elFarto, 2025-02-18:
I've flashed this firmware, configured for modbus v3.05 to a ShineWIFI-S stick, and it seems to be working fine. However yesterday I noticed some odd temperature readings:

[image]

The temperature gets to 29.7C, then drops down to 25.5C but seems to still act normally except being ~4C lower than it should be. I thought it might be some integer overflow issue at first, but that doesn't quite explain what's happening.

Does anyone know what's going on here?

elFarto, 2025-03-01:
It seems the issue gets weirder:

[image]

It's quite capable of getting past 29.7C once it's past that temperature once. I'm at quite a loss to explain it. I've looked through the code that manipulates the values, but I don't see any way it could cause this sort of issue.

It seems the first time the temperature register tries to get to 300, it then gets 45 subtracted from it, until it passes below 300 again.

crasu, 2025-05-07:
Does you inverter have a fan?

closed 2025-05-07 by crasu