OpenInverterGateway/OpenInverterGateway #464: AP not showing up after flashing

hackex, 2025-02-18:
I have a Shine-S Stick from Growatt and used tasmota and esphome web flasher to flash a precompiled OpenInverterGateway. After taking away the connector from GND to gpio0, all the LEDs are showing up but no Growatt AP is opened to connect to.

Anyone know what I can try to solve this issue?

0000vt, 2025-02-18:
I just flashed as well, had to hold the button on the wifi-x stick till the blue started blinking. Then i got the AP

hackex, 2025-02-18:
doesn't work on my Shine-S

ChrisHVL, 2025-02-18:
Are you using a FTDI-Adapter?
These adapters are sometimes a bit triky or contain not the original chip, which let them only work half.
I had some "fun" on using them.

hackex, 2025-02-18:
but those Adapters should work just to get the right voltage to power the stick, correct?

hackex, 2025-02-18:
And if ESPhome web shows that the flashing was complete, I think the writing process was, indeed, successful

hackex, 2025-02-18:
ok. It is really the Adapter. From the attached picture showing all adapters, I have tried...only the top one worked with the 3.3V
Thanks

[image]

hackex, 2025-02-18:
Is there any way to see the current installed version of OpenInverter firmware in the menue?

ChrisHVL, 2025-02-18:
Normally I is shown when you are using a original Growat stick. You have to enter into setup where you need a password which you can find in the internet. Be careful not to change values, you cannot judge.

73

ChrisHVL, 2025-02-18:
The password is something with todays date and growatt I do remember.